magicbug/Cloudlog #1703: Unable to post QSO with TO2DL

kconkas, 2022-10-19:
**Describe the bug**
I am unable to post QSO with station TO2DL dating 19-10-2022 at 11:49 on 12 m using CW with reports 599 both ways

**To Reproduce**
Steps to reproduce the behaviour:
1. From QSO menu select "Post QSO"
2. Fill in QSO details as above
3. Click "Save QSO"

**Expected behaviour**
QSO is saved and visible under "previous contacts".

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: MacOS Monterey 12.5
 - Browser: Google Chrome
 - Version: 105.0.5195.102

**Smartphone (please complete the following information):**
 - Device: Samsung Galaxy S21 5G
 - OS: Android 12
 - Browser: Firefox 
 - Version: 106.1.0

**Additional context**
I am still on cloudlog v1 commit ed33967472 (but will upgrade to v2 promptly if this behaviour is addressed in v2).

The behaviour doesn't seem to be browser or OS specific - I'm experiencing identical issue on several devices I tried. I suspect the issue is with the call sign itself as I am able to post other QSOs. 

In developer console I see 2 errors each time I click "Save QSO":
- "An invalid form control with name='dxcc_id' is not focusable."
- "An invalid form control with name='cqz' is not focusable. "

[image]

dg9vh, 2022-10-19:
Try update the dxcc list

kconkas, 2022-10-19:
Thanks @dg9vh, this solved my issue.

closed 2022-10-19 by kconkas