magicbug/Cloudlog #1710: Dropdown menu in search results doesn't scroll properly (clublog v2)

kconkas, 2022-10-22:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
After searching for a station, individual QSOs with it get listed and to the right of each there is a cogwheel icon with options to to edit QSO, mark QSLs as received/sent etc. This menu doesn't scroll so it is impossible to click on anything other than the top 3 menu items.

**To Reproduce**
Steps to reproduce the behaviour:
1. In "Search Callsign" box type a call sign and hit enter
2. Locate a QSO of interest and click the cogwheel button

**Expected behaviour**
I can see all menu options.

**Screenshots**
[image]

Screen recording:
https://user-images.githubusercontent.com/8178837/197344534-c266e805-3847-4ba1-b9a9-fd601f0848b3.mov

**Desktop (please complete the following information):**
 - OS: MacOS Monterey 12.6
 - Browser Firefox and Google Chrome
 - Version Firefox 104.0.1 and Google Chrome 106.0.5249.119

**Additional context**
I believe I may have got this menu to scroll correctly only once but I am unable to reproduce it even after logging out of clublog, restarting either browser or trying in incognito mode.

AndreasK79, 2022-10-22:
Created #1712 with a fix that works on my setup:
[image]

AndreasK79, 2022-10-27:
This has been fixed in dev.