magicbug/Cloudlog #1711: Uploaded LoTW certificate isn't shown

kconkas, 2022-10-22:
**Describe the bug**
When I upload a LoTW certificate, the upload seems to complete successfully but I can't see the newly uploaded certificate and there's still the "You need to upload some LoTW p12 certificates to use this area." message in "Available Certificates"

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to "Logbook of the World"
2. Click on "Upload certificate"
3. Select .p12 file and DXCC country
4. Click "Upload File"

**Expected behaviour**
The freshly uploaded certificate is listed under "Available Certificates".

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: MacOS Monterey 12.6
 - Browser Firefox, Google Chrome
 - Version Firefox 104.0.1, Google Chrome 106.0.5249.119

**Additional context**
I can see the uploaded files under `uploads/lotw/certs`:
```
kconkas@homeassistant:/var/www/html/cloudlog$ ls -la uploads/lotw/certs/
total 24
drwxr-xr-x 2 www-data www-data 4096 Oct 21 23:34 .
drwxr-xr-x 3 www-data www-data 4096 Jul 27 01:18 ..
-rw-r--r-- 1 www-data www-data 6725 Oct 21 23:34 M0NKC1.p12
-rw-r--r-- 1 www-data www-data 6725 Oct 21 23:32 M0NKC.p12
```
I tried this both with leaving `application/config/lotw.php` unmodified as well as setting `$config['lotw_keys_folder']` in it to a location outside www root, all to no avail.

magicbug, 2022-10-24:
Also confirm that the p12 isn't password protected as Cloudlog will have issues with that

kconkas, 2022-10-24:
@phl0 with `log_threshold` set to 2 I get the following in the log:
```
DEBUG - 2022-10-24 21:56:29 --> cURL Class Initialized
DEBUG - 2022-10-24 21:56:37 --> cURL Class Initialized
ERROR - 2022-10-24 21:56:38 --> error:0308010C:digital envelope routines::unsupported
DEBUG - 2022-10-24 21:56:38 --> cURL Class Initialized
```
@magicbug I am not using a password for my .p12 file.

magicbug, 2022-10-24:
Can you explain the host setup?

kconkas, 2022-10-25:
Sure:
- OS: Ubuntu 22.04.1 LTS
- Web server: Apache 2.4.54-1
- PHP: 7.4.32

The full list of installed PHP 7.4 modules is as follows:
```
kconkas@homeassistant:~$ dpkg -l | grep php
ii  libapache2-mod-php7.4                         1:7.4.32-1+ubuntu22.04.1+deb.sury.org+2    amd64        server-side, HTML-embedded scripting language (Apache 2 module)
ii  php-common                                    2:92+ubuntu22.04.1+deb.sury.org+1          all          Common files for PHP packages
ii  php7.4                                        1:7.4.32-1+ubuntu22.04.1+deb.sury.org+2    all          server-side, HTML-embedded scripting language (metapackage)
ii  php7.4-cli                                    1:7.4.32-1+ubuntu22.04.1+deb.sury.org+2    amd64        command-line interpreter for the PHP scripting language
ii  php7.4-common                                 1:7.4.32-1+ubuntu22.04.1+deb.sury.org+2    amd64        documentation, examples and common module for PHP
ii  php7.4-curl                                   1:7.4.32-1+ubuntu22.04.1+deb.sury.org+2    amd64        CURL module for PHP
ii  php7.4-json                                   1:7.4.32-1+ubuntu22.04.1+deb.sury.org+2    amd64        JSON module for PHP
ii  php7.4-mbstring                               1:7.4.32-1+ubuntu22.04.1+deb.sury.org+2    amd64        MBSTRING module for PHP
ii  php7.4-mysql                                  1:7.4.32-1+ubuntu22.04.1+deb.sury.org+2    amd64        MySQL module for PHP
ii  php7.4-opcache                                1:7.4.32-1+ubuntu22.04.1+deb.sury.org+2    amd64        Zend OpCache module for PHP
ii  php7.4-readline                               1:7.4.32-1+ubuntu22.04.1+deb.sury.org+2    amd64        readline module for PHP
ii  php7.4-xml                                    1:7.4.32-1+ubuntu22.04.1+deb.sury.org+2    amd64        DOM, SimpleXML, XML, and XSL module for PHP
```
There is also PHP 8.1 installed on the system but I am pretty sure I am not using it for cloudlog:
```
kconkas@homeassistant:/etc/apache2$ cat mods-enabled/php7.4.load
# Conflicts: php5
# Depends: mpm_prefork
LoadModule php7_module /usr/lib/apache2/modules/libphp7.4.so
kconkas@homeassistant:/etc/apache2$ cat mods-enabled/php7.4.conf
<FilesMatch ".+\.ph(ar|p|tml)$">
    SetHandler application/x-httpd-php
</FilesMatch>
<FilesMatch ".+\.phps$">
    SetHandler application/x-httpd-php-source
    # Deny access to raw php sources by default
    # To re-enable it's recommended to enable access to the files
    # only in specific virtual host or directory
    Require all denied
</FilesMatch>
# Deny access to files without filename (e.g. '.php')
<FilesMatch "^\.ph(ar|p|ps|tml)$">
    Require all denied
</FilesMatch>

# Running PHP scripts in user directories is disabled by default
#
# To re-enable PHP in user directories comment the following lines
# (from <IfModule ...> to </IfModule>.) Do NOT set it to On as it
# prevents .htaccess files from disabling it.
<IfModule mod_userdir.c>
    <Directory /home/*/public_html>
        php_admin_flag engine Off
    </Directory>
</IfModule>
```

phl0, 2022-10-25:
TBH: Looks like some OpenSSL stuff is missing. I am trying to find out what exactly.

magicbug, 2022-10-25:
Certainly does look like an OpenSSL issue

kconkas, 2022-10-25:
It may as well be, but I don't know what I may be missing. PHP seems to have ssl support compiled in:
```
kconkas@homeassistant:/etc/apache2$ php -i  | grep ssl
Registered Stream Socket Transports => tcp, udp, unix, udg, ssl, tls, tlsv1.0, tlsv1.1, tlsv1.2, tlsv1.3
libSSH Version => libssh/0.9.6/openssl/zlib
openssl
Openssl default config => /usr/lib/ssl/openssl.cnf
openssl.cafile => no value => no value
openssl.capath => no value => no value
```
And openssl is also installed:
```
kconkas@homeassistant:~$ dpkg -l | grep openssl
ii  openssl                                       3.0.2-0ubuntu1.6                           amd64        Secure Sockets Layer toolkit - cryptographic utility
ii  perl-openssl-defaults:amd64                   5build2                                    amd64        version compatibility baseline for Perl OpenSSL packages
```

kconkas, 2022-10-25:
Gents, you were spot on. I googled the error and found this article: https://stackoverflow.com/questions/73832854/php-openssl-pkcs12-read-error0308010cdigital-envelope-routinesunsupported The 2nd solution (with enabling SSL legacy provider) worked for me. I just wanted to include this information in case someone else encounters the same issue. Thank you for your help. I am closing this issue now.

closed 2022-10-25 by kconkas

phl0, 2022-10-26:
Well I read about this in other contexts. My environment ist similar (if not identical) to yours but I did not encounter this issue (and did noch enable the legacy provider). Maybe my certificate is different regarding the technical details ...