magicbug/Cloudlog #1720: Pagination in public view

PD0RZH, 2022-10-26:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behaviour:
1. Make sure you've got enough QSO's  for pagination
2. Go to Public view 
3. Scroll down to  bottom
4. No pagination

**Expected behaviour**
Pagination to go through all QSO's

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser  Edge
 - Version 106.0.1370.52 (Officiële build) (64-bits)

And if it's not a bug then maybe a feature request

magicbug, 2022-10-26:
scratch that its a bug! seems public has no logbook.. I'll create a request

PD0RZH, 2022-10-26:
Serioously i was not sure whether this isa bug or not. Glad you created a request.

patrickrb, 2024-04-01:
I think public logbooks are working again.  Should this one be closed? Or is there more work to be done?

patrickrb, 2024-04-01:
Also, i was talking to my club about this today.  We think itd be nice if we could share our entire log publicly.  Would it be ok if we added an option in the logbook options to 'allow full access'? Or something to that affect, that would allow paging/search etc through the entire log publicly?

closed 2024-04-04 by magicbug

PD0RZH, 2024-04-04:
Are you sure pagination is working in public view? 
[image]

cloudlog/index.php/visitor/logbook

PR #3045 (2024-04-04): Visitor page pagination
Closes #1720 

- Added pagination to visitors page ~*only* if the public search option is enabled~
- ~Happy to update this PR if you'd rather a different config option or any other feedback, but to me it felt like if search is enabled then pagination might make sense too.  My biggest concern would be that people who have this feature enabled currently might not want the paginated results.~

Enhancements to public search functionality:

* [`application/controllers/Visitor.php`](diffhunk://#diff-e5390d77eb91cd0945b808102973347ba151fe27ed0580d54cfb99cab6a3bfb1R78-R109): In [... cut]