magicbug/Cloudlog #1743: SOTA CSV Export does not interpret QSOs correctly

florianprischl, 2022-11-06:
**Describe the bug**
The SOTA CSV Export does not produce a correct list of QSOs for the given parameters. It appears that the generated CSV file switches the "other station" and "own station" SOTA reference. As a result, CSV files imported to https://www.sotadata.org.uk do not result in correct logs.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to Station Menu (top right) --> 'SOTA CSV Export' --> Choose relevant parameters to generate logs for SOTA
2. Click on 'Export', download the CSV file.
3. Go to https://www.sotadata.org.uk
4. Go to 'Menu' (top left of the page) --> 'Submit Logs' --> 'Upload CSV/TSV log file' --> Chose your previously downloaded CSV file
5. See error on the import confirmation page: Only QSOs that include a SOTA reference for the other station will be shown.

Additional symptom: The CSV file actually only includes the wrong QSOs. The issue is obviously not in the formatting of the CSV file, but what cloudlog includes in the CSV file in the first place.

**Expected behaviour**
The CSV file should include all QSOs following the parameters chosen on the CSV Export page.

**References**
Check out the following files which helped me diagnose the problem:

1. This is an erroneous CSV file. It only includes 1 QSO even though it should include 4: https://oe1flp.at/files/OE1FLP-SOTA-06112022-1935.csv
2. This is the ADIF file for the same station and the same day, created through cloudlog's ADIF Export feature: https://oe1flp.at/files/OE1FLP-06112022-1938.adi

The ADIF file creates a correct log upon import on https://www.sotadata.org.uk/, the CSV file does not.

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro 21H2
 - Browser Google Chrome
 - Version 107.0.5304.88

**Additional context**
Running self-hosted cloudlog 2.0, fully updated.

phl0, 2022-11-07:
Hmm think I built this and uploaded QSOs correctly several times. That was chaser logs only. I will have a look. @magicbug please assign to me.

phl0, 2022-11-07:
I did a test export from my Cloudlog instance. The result is:

[image]

Which from my point of view is inline with the example file on https://www.sotadata.org.uk/en/upload/activator/csv/info:

[image]

phl0, 2022-11-07:
By the way I just uploaded your csv file to my SOTA account successfully ... (and deleted it again)...

[image]

[image]

florianprischl, 2022-11-07:
> I did a test export from my Cloudlog instance. The result is:
> 
> [image]
> 
> Which from my point of view is inline with the example file on https://www.sotadata.org.uk/en/upload/activator/csv/info:
> 
> [image]

I agree that the CSV is formatted correctly, apart from the fact that it uses semicolons instead of commas (but the SOTA database import accepts it without a problem). The CSV file is not the problem.

The problem appears to lie with the export from Cloudlog to the CSV. Namely, Cloudlog does not export QSOs which contain a SOTA reference for your own station but do not contain a SOTA reference for the other station (i.e. non-S2S QSOs).

florianprischl, 2022-11-07:
> By the way I just uploaded your csv file to my SOTA account successfully ... (and deleted it again)...
> 
> [image]
> 
> [image]

Right, because the CSV only contains this one QSO. It should, however, contain four QSOs. Compare this with the ADIF file I provided, which correctly includes four QSOs with the SOTA reference for my station.

You will notice that the one QSO retained in the CSV is an S2S entry. It includes SOTA references both for the submitting station as well as the other station. The other three QSOs were regular, non-S2S QSOs and are not included in the CSV.

Thank you for your diligence and work in pursuing this!

mrmabs, 2022-11-08:
I just tried to generate a SOTA Activator log and I get zero bytes. Digging a bit further, I was able to generate a log file from another activation, and the only entry in the file was someone I contacted S2S, so they had a SOTA reference.

From what I can infer, "My Summit" (from Station Location) is not being checked (to mark the QSO as SOTA for inclusion) or included in the generated log file.

phl0, 2022-11-08:
Roger I will check and fix that.

phl0, 2022-11-08:
Issue confirmed and issued PR https://github.com/magicbug/Cloudlog/pull/1747 to fix that.

closed 2022-11-09 by magicbug