magicbug/Cloudlog #1761: SOTA CSV Export does not honor export filters

florianprischl, 2022-11-09:
**Describe the bug**
The SOTA CSV export does not respect filters and instead exports all QSLs that have a SOTA reference.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to main menu (top right)
2. Click on 'SOTA CSV Export'
3. Select filter criteria which you know should provide only a subset of SOTA QSOs (e.g. only from one specific location or in a specific date range).
4. Click 'Export' and check the resultant CSV file.

**Expected behaviour**
Cloudlog SOTA CSV export should create a CSV in compliance with the filters set by the user.

**Screenshots**
These are the filter settings I have used for the demonstration:
[image]

And this is the resultant CSV file:
`V2,OE1FLP/M,,27/07/22,0820,144MHz,FM,OE5REO/P,OE/OO-050,SOTA activator`
`V2,OE1FLP/M,,27/07/22,0842,144MHz,FM,OE5HCE/P,OE/OO-163,SOTA activator`
`V2,OE1FLP/5,,29/08/22,1327,144MHz,FM,OE5FFE/P,OE/OO-012,`
`V2,OE1FLP/P,OE/SB-044,10/09/22,1155,144MHz,FM,OE3WHU/P,OE/OO-129,`
`V2,HA/OE1FLP/P,HA/ND-001,18/09/22,1033,144MHz,FM,OE1CCX/P,OE/NO-149,S2S`
`V2,HA/OE1FLP/P,HA/ND-001,18/09/22,1052,144MHz,FM,OM1WS,OM/BA-004,`
`V2,OE1FLP/P,OE/NO-204,06/11/22,1242,144MHz,FM,OE1WED/3,OE/NO-279,`
`V2,OE1FLP/P,OE/NO-204,06/11/22,1245,144MHz,FM,OE3GJC,,`
`V2,OE1FLP/P,OE/NO-204,06/11/22,1247,144MHz,FM,OE1CCX/P,,`
`V2,OE1FLP/P,OE/NO-204,06/11/22,1247,144MHz,FM,OE6FEF,,`

Note that this includes entries from a) station locations other than the one I set with the filter (OE/NO-204) and b) dates outside the selected date range (1-6 November 2022).

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro 22H2
 - Browser Google Chrome
 - Version 107.0.5304.107 (Official Build) (64-bit)

**Additional context**
Version 2.1 included commit ae552f2dda2e0f8c63abe05ffb4f196dae1a9ec5, which fixed #1743. I believe that either this commit or possible one of the other commits submitted by @phl0 in the course of fixing #1743 created the bug. Unfortunately I don't know PHP so I can't really contribute.

phl0, 2022-11-10:
Please assign to me.

magicbug, 2022-11-10:
> Please assign to me.

Your wish is my command 😅

phl0, 2022-11-10:
Confirmed. And fixed with https://github.com/magicbug/Cloudlog/pull/1769.