magicbug/Cloudlog #1767: Can't create a location

ruilvo, 2022-11-10:
On master branch as of today, when I try to create a station location, I can't select a DXCC entity:
[image]

On `production` mode, nothing happens. On `developer` mode, this error pops up:
[image]

Confirming it is indeed due to the inability to select the DXCC entity.

This is how I setup the server, it's reproduceable for anybody with docker: https://github.com/ruilvo/cloudlog-docker-compose

phl0, 2022-11-10:
Did you update the DXCC information?

ruilvo, 2022-11-10:
That was it, thanks.

closed 2022-11-10 by ruilvo