magicbug/Cloudlog #1777: New proposed option for customizable logbook fields

ofadam, 2022-11-14:
As someone who regularly activates Parks On The Air sites, I'd love to be able to customize one of the 5 logbook columns to show Station Location.

One potential issue is that the name could be rather long... which would. Two possible solutions could be just using ellipses after a certain number of characters (with the user expected to keep names short, with distinguishable info at the front) or maybe a "short name" field in the Station Location setup?

phl0, 2022-11-14:
This could be possible as soon as ADIF supports separate fields for POTA. This has been proposed and will be added with 3.1.4. But thats not been released yet. I have a Cloudlog patch ready for the day it comes. 😇

https://github.com/phl0/Cloudlog/tree/POTA

magicbug, 2022-12-21:
@phl0 did POTA field get added to the field selection cant honestly remember since we committed the code

phl0, 2022-12-21:
> @phl0 did POTA field get added to the field selection cant honestly remember since we committed the code

I think I misunderstood the request here. The op wants to show his own POTA location as user definable column I guess. I didn't add that.

phl0, 2022-12-21:
@ofadam Should this show your POTA location or the one of the DX station?

closed 2022-12-26 by magicbug