magicbug/Cloudlog #1783: two active stations

DL3EL, 2022-11-15:
After the upgrade to Cloudlog 2.0 yesterday I have two active stations. I do not know, how I managed to do that. It is not possible to get it inactive for deletion. Any ideas?

See screenshot.

73, Thomas, DL3EL

**Desktop (please complete the following information):**
 - OS: [Raspian Buster]
[image]

phl0, 2022-11-15:
Thats not an expected result :D What happens if you activate another station?

magicbug, 2022-11-15:
Definitely a bug, fastest method to fix would be to edit the MySQL table station_profile and find anything with an "Station_Active" then set them to null, once you do this you'll need to set a location to active in the GUI to get things working correctly again.

If your not comfortable can make a tool to fix this later :)

DL3EL, 2022-11-15:
> Definitely a bug, fastest method to fix would be to edit the MySQL table station_profile and find anything with an "Station_Active" then set them to null, once you do this you'll need to set a [...]

will try, can you point my to the location if the table?

phl0, 2022-11-15:
The table name is `station_profile`. Try something like:

```
UPDATE `station_profile` SET `station_active` = NULL WHERE `station_id` = 7;
```

DL3EL, 2022-11-15:
thanks, that has worked. Problem solved..

closed 2022-11-15 by magicbug

phl0, 2022-11-15:
I made a fix to prevent such in the future. See https://github.com/magicbug/Cloudlog/pull/1786.