magicbug/Cloudlog #1787: minor issue, re colors (no bug)

DL3EL, 2022-11-15:
as I am colorblind, I have bigger difficulties to differenciate between the red and green arrows in the log display. As a personal hack, I have altered log_ajax.php, to show the received eqsl_arrow in yellow. 

That is bad, because leads to issues with the next git pull (and that doesn't work for paper qsl, for whatever reason).

Maybe there could be an easier way in the future to alter the colours a bit :-)

vy 73 de Thomas, DL3EL

AndreasK79, 2022-11-15:
Need to take a closer look, but seems classes are set, like qsl-red. Must test, but I think it means that these can be set in overrides.css in the theme folder.

phl0, 2022-11-15:
That was also my idea :)

AndreasK79, 2022-11-15:
[image]

AndreasK79, 2022-11-16:
eQSL received is done a bit differently. The style is set directly. Might be because you can click on it and see eQSL. I'll see if that is an easy fix.

DL3EL, 2022-11-16:
Thanks. That is correct, I had a look at log_ajax.php where I saw that (and changed for a test. My suggestion for a quick fix would be to change in Line 183 the style into a class:
<a class="eqsl-green" href="<?php echo site_url("eqsl/imag ......

works for me

AndreasK79, 2022-11-16:
It didn't look quite right when I tried, but this one worked:
`<a style="text-decoration:none" href="<?php echo site_url("eqsl/image/".$row->COL_PRIMARY_KEY); ?>" data-fancybox="images" data-width="528" data-height="336"><span class="eqsl-<?php echo ($row->COL_EQSL_QSL_RCVD=='Y')?'green':'red'?>">&#9660;</span></a>`

Will push fix later.

Also I was thinking, that you might want to copy the default theme to another theme directory and add it to Cloudlog. That way, git will not complain that overrides.css has been modified.

phl0, 2022-11-16:
> Btw.: when I set public search bar to "enabled", my expectation would be to see a logbook, when connecting without a valid login. That is not the case. It always starts with http://192.168.241.99/index.php/user/login

See https://github.com/magicbug/Cloudlog/issues/1505.

AndreasK79, 2022-11-16:
@DL3EL did you set the theme in the user account? The one under options, is just for public access.

phl0, 2022-11-16:
https://github.com/magicbug/Cloudlog/pull/1792 is the fix for the color stuff with eqsl.

phl0, 2022-11-16:
@DL3EL To make it permanent and not clash with future updates I would clone the default theme as @AndreasK79 mentioned, add your overrides there and use that theme. Can be done with:

```
cp -prv assets/css/default assets/css/default_altColor
```
and some SQL:
```
INSERT INTO `themes` (`id`, `name`, `foldername`) VALUES (NULL, 'default (alt colors)', 'default_altColor');
```

After refreshing you should be able to select the default (alt colors) theme.

DL3EL, 2022-11-16:
Thanks, I did that earlier today. But after doing that, I noticed, that you do not need to enter the SQL statements on the Servers CLI. You can use Admin/Themes in the WebGui for that...

phl0, 2022-11-16:
Thank you guys. So even I learned something from this :)

AndreasK79, 2022-11-16:
I don't mind at all @phl0  I didn't have time to fix anything right now anyway.

As for features... the wiki is lacking. More fun to make stuff than document :)

magicbug, 2022-11-16:
If any non coders want to help with the docs then help is always accepted

DL3EL, 2022-11-16:
Peter,
as I am more a coder in Perl (https://relaislisten.darc.de/) or in C++ (https://github.com/DL3EL/TTGO-T-Beam-LoRa-APRS) I will contribute here in helping with the docs. Any specific things which are urgent? Otherwise I would look through the wiki and add missing things in then next couple of days. 
Thomas

magicbug, 2022-11-20:
As this is fixed in dev branch going to close docs discussions welcome

closed 2022-11-20 by magicbug