magicbug/Cloudlog #1798: Ancillary suffixes not handled correctly

kconkas, 2022-11-19:
**Describe the bug**
Ancillary suffixes, for station operating outside its home country using a suffix to identify such operation, are ignored and the country is logged incorrectly as the home country. Example: `VE1CWJ/VP9` is logged as "Canada" instead of "Bermuda".

**To Reproduce**
Steps to reproduce the behaviour:
1. In "QSO" menu click "Post QSO"
2. In "Callsign" textbox enter affected call sign (for example "VE1CWJ/VP9")

**Expected behaviour**
Country is derived from ancillary suffix.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: MacOS Monterey 12.6
 - Browser: Firefox 104.0.1

magicbug, 2022-11-20:
Aware fix in progress, temp solution is just select the correct dxcc for now

dg9vh, 2022-11-20:
Should that not be logged as VP9/VE1CWJ?

73 de Kim
DG9VH

magicbug, 2022-11-20:
North America put it on the end rather than front.

dg9vh, 2022-11-20:
Ah, ok.

kconkas, 2022-11-20:
> Should that not be logged as VP9/VE1CWJ? 73 de Kim DG9VH Kristijan Conkas ***@***.***> schrieb am So., 20. Nov. 2022, 00:44:
Both forms are valid. Also, I can only log the call sign as it is given on air or else I may have problems when trying to confirm the QSO.

AndreasK79, 2022-12-29:
This has been fixed in the dev branch.

closed 2022-12-29 by AndreasK79