magicbug/Cloudlog #1799: eQSL Download  - "It seems that the eQSL site has changed...!

ThomasDL3NGN, 2022-11-21:
**Describe the bug**
Download of eQSL QSOs did not work. CloudLog displays "It seems that the eQSL site hase changed. Pleas open up a issue on GitHub."

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to "User"
2. Click on "eQSL Import/Export"
3. Select "Import directly from eQSL"
4. Pusch button "Import the eQSL Matches" 

**Expected behaviour**
eQSL matches should marked as QSL received in CloudLog

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Chromium Version 92.0.4492.0 / Firefox Version 107.0

phl0, 2022-11-21:
Hhhmmm, mine is still working well...

ThomasDL3NGN, 2022-11-21:
I tried to switch of Pi-hole -> still the same here...
"It seems that the eQSL site has changed. Please open up an issue on GitHub."

Sorry... I have no idea...

magicbug, 2022-11-21:
This is likely to be an issue with your username/pass (user account) and the eQSL QTH Nickname (Station Location config) I'd double check those are correct

closed 2022-11-26 by magicbug