magicbug/Cloudlog #1800: API not updating frequency and mode

TriBallscom, 2022-11-21:
After upgrading to Cloudlog 2.1 (comming from 1.x) the API call does not update the frequency and mode.
The name of the radio is registered so the api call does work

Tested with:

curl --silent --insecure \
         --header "Content-Type: application/json" \
         --request POST \
         --data "{
           \"key\":\"xxxxxxxxxxxxx\",
           \"radio\":\"TestRadio\",
           \"frequency\":\"1838999\",
           \"mode\":\"ssb\",
           \"timestamp\":\"$(date -u +"%Y/%m/%d %H:%M")\"
         }" https://<url to cloudlog>/index.php/api/radio

phl0, 2022-11-21:
Tested ok here with some escapes:

```
curl --insecure --header "Content-Type: application/json" --request POST --data "{ \"key\":\"xxxxxxxxxxx\", \"radio\":\"TestRadio\", \"frequency\":\"1838999\", \"mode\":\"ssb\", \"timestamp\":\"$(date -u +"%Y/%m/%d %H:%M")\" }" https://host.name/cloudlog/index.php/api/radio
```

[image]

[image]

TriBallscom, 2022-11-21:
> Any hints in the API response on the console if you omit "--silent"?

No just a 
{"status":"success"}

TriBallscom, 2022-11-21:
> Tested ok here with some escapes:

tested the same but same result.
[image]
[image]

phl0, 2022-11-21:
> [image]

So frequency is set here. Cannot see an error?!

phl0, 2022-11-21:
> [image]

Frequency is set. Cannot see an error here?!

TriBallscom, 2022-11-21:
> Frequency is set. Cannot see an error here?!

It's not the frequency I parsed to the api

phl0, 2022-11-21:
> It's not the frequency I parsed to the api

Oh right. I overlooked the decimals... Will have a look.

phl0, 2022-11-21:
It sets the freqency here correctly up to the single Hz. 

[image]

The JSON response also contains the exact frequency as above.

So only possible thing I can think of currently is the browser. Which one are you using?

TriBallscom, 2022-11-21:
> So only possible thing I can think of currently is the browser. Which one are you using?

I checked with Chrome and Edge on Windows 11.
It's weird that the variable is set in the "hardware interfaces" but not in the Live QSO screen.

TriBallscom, 2022-11-21:
[image]

magicbug, 2022-11-21:
I think this is a simple case of Cloudlogs expecting the freq in Hz via the API

phl0, 2022-11-21:
> [image]

Can you check the API JSON response for the value the API gives back?

TriBallscom, 2022-11-22:
> Can you check the API JSON response for the value the API gives back?

The API works, in the "hardware interfaces" it gives the correct frequency just not on the Live QSO page

WaarlandIT, 2022-11-27:
@phl0 What browser are you using? 
I try to replicate the screenshots you made but my views are different and to make sure we look at the same thing I like to use the same browser you are using.

phl0, 2022-11-27:
I am using Firefox on Linux (Ubuntu) as default.

WaarlandIT, 2022-11-27:
Interesting.. You get a json response and I don't .. 
I send a SSB Mode to the API and in the hardware interfaces page that is shown but in the Live QSO page I only get an AM and SAT mode but no SSB. The AM mode shows the frequency shown on the page as default 

[image]

TriBallscom, 2022-11-27:
I just realized I commented with the wrong account ;) WaarlandIT, is another account of mine..

phl0, 2022-11-27:
> I send a SSB Mode to the API and in the hardware interfaces page that is shown but in the Live QSO page I only get an AM and SAT mode but no SSB. The AM mode shows the frequency shown on the page as default

Can you show the curl request? I will try to reproduce here.

TriBallscom, 2022-11-27:
Sure.. just removed the hostname.domain and the api key.. 

curl --insecure --header "Content-Type: application/json" --request POST --data  "{\"key\":\"XXXXXXXXXXXX\", \
\"radio\":\"QO-100 Station\",
\"frequency\":\"24001700\",
\"mode\": \"SSB\",
\"timestamp\":\"2022/11/26 10:47\"}" https://XXXXXXXXXXXXXXXXXX/index.php/api/radio

TriBallscom, 2022-11-27:
extra info I use Ubuntu 20.04 with PHP7.4

TriBallscom, 2022-11-27:
Ok some new development in my research. 
I installed Cloudlog on a Ubuntu 18.04 with PHP 7.2 and now it works. 
Not sure what the difference is but it is interesting

TriBallscom, 2022-11-27:
If I look at the hardware interfaces page I see that the notation of the frequency is also different when running on Ubuntu 18.04 with php7.2
[image]

phl0, 2022-11-27:
> If I look at the hardware interfaces page I see that the notation of the frequency is also different when running on Ubuntu 18.04 with php7.2 [image]

Seems you missed to pass the frequency in Hz on the API request?

TriBallscom, 2022-11-27:
> Seems you missed to pass the frequency in Hz on the API request?

No same request from my workstation just updated the timestamp

phl0, 2022-11-27:
> No same request from my workstation just updated the timestamp

Roger. I will try to debug ...

TriBallscom, 2022-11-27:
Just to be sure.. What are the specs of your server? Is that Ubuntu 18.04?

phl0, 2022-11-29:
> Just to be sure.. What are the specs of your server? Is that Ubuntu 18.04?

Sri delay. Its Ubuntu 20.04. Apache 2.4.41. PHP 7.4.3.

TriBallscom, 2022-11-30:
That was my setup when it failed.. weird.

closed 2022-12-26 by magicbug