magicbug/Cloudlog #1809: Nearly empyty qso reappears in `TABLE_HRD_CONTACTS_V01` and to be uploaded to eqsl

daccle, 2022-11-26:
**Describe the bug**
When I want to upload to esql (https://<cloudlog-url>/eqsl/Export) a log entry with on call appears repeatedly. 

Date | Call | Mode | Submode | Band | eQSL QTH Nickname
2021-05-29 20:13:30 |   | FT8 |   | 20m |  
2021-05-29 20:13:30 |   | FT8 |   | 20m |  
2021-05-29 20:13:30 |   | FT8 |   | 20m |  

Last time I deleted circa, 7000 of such entries from the DB:

DELETE FROM `TABLE_HRD_CONTACTS_V01` WHERE `COL_QSO_DATE` = '2021-05-29' AND `COL_CALL` = ''

But these entries reappear each time I visit https://<cloudlog-url>/eqsl/Export. When I just stay on this page, more and more such entries appear in the DB.

**Expected behaviour**
I don't want these enrtries to appear. 

**Desktop (please complete the following information):**
 - OS: ANY
 - Browser Firefox
 - Version 107

magicbug, 2022-11-26:
After deleting the empty recordings are you reuploading the FT8 QSOs in the background so they are just getting pushed back in? although not sure how they are getting added

daccle, 2022-11-26:
Thank you! Just deleted the corresponding API key and voilà! Sorry for bothering you.

closed 2022-11-26 by daccle

magicbug, 2022-11-26:
I would suggest asking the developer of the WSJT application to not allow saving ADIFs with incomplete data.

daccle, 2022-11-26:
Good point! But I guess these were the remains of my own wsjt-adiflog-watcher attempts, which I've totally forgotten about...