magicbug/Cloudlog #1810: Upload to clublog fails with no intelligible error message

daccle, 2022-11-26:
**Describe the bug**
I am not able to upload to clublog anymore.  When I visit https://<cloudlog-url>/clublog/upload/<call> I only get the message Error. In the log there is written:  

ERROR - 2022-11-26 15:09:25 --> Clublog upload for MYCALL failed reason 

**Expected behaviour**
I want to be able to upload to clublog or at least have a meaningful error message

magicbug, 2022-11-26:
I believe the error is what's returned by Clublog we are of course not responsible for their system and can only show what they provide.

as per Cloudlogs documentation calling the clublog/upload must also include the username for the account which currently holds the login information /clublog/upload/<username> this will be changed soon but that's how it works

daccle, 2022-11-26:
Not sure about that, this is a clublog error. I tried to log the response from the curl statement in controllers/Clublog.php:105, and it is empty. How can I check if the call was really made. Maybe my PHP process is not allowed to do so?

daccle, 2022-12-09:
While I wasn't able to understand what exactly has happend, I was able to recover, by manually exporting and uploading the missings qsos, deleting the clublog<hash>.adi files in /upload and marking the qsos as exported in the DB.

closed 2022-12-09 by daccle