magicbug/Cloudlog #1811: Bug: Gridsquare Analytics is not aware of Band at the 1st usage

int2001, 2022-11-27:
When you go to Analytics --> Gridsquare and choose "Band", the Dropdown is (on the 1st usage) on "All".
But Analytics are applied to 2m-Band only.

phl0, 2022-11-27:
Hmm in my case "2m" is selected upon clicking "Band". See link in the lower left of the screenshot.

[image]

int2001, 2022-11-28:
Ok, i try to specify it.
When you click on "Band", 2m is shown, but the Dropdown shows "All".
Solution:
a) in Dropdown there's "2m" selected
b) the "All"-Page ist rendered

i'd prefer "b". At the moment this is inconsistent

[image]

phl0, 2022-11-28:
> Ok, i try to specify it. When you click on "Band", 2m is shown, but the Dropdown shows "All". Solution: a) in Dropdown there's "2m" selected b) the "All"-Page ist rendered

I understand. Just tried to explain that I could not reproduce it. What browser are you using?

int2001, 2022-11-28:
Problem exists with
- Chrome (107.0.5304.110)
- Current Chrome on iOS
- Safari (16.1 (18614.2.9.1.12))
- Current Safari on iOS

phl0, 2022-11-28:
Seems like it is affecting only specific browsers. I tried with Firefox, Chromium and Safari on an iPhone but could not reproduce the issue.

int2001, 2022-11-28:
Thats interesting. 
In the (rendered) js-Annex of the Page, 2m are missing. That way the page couldn't mark the option as selected. (Doesn't exist) - take a look at the Screenshot

So i took a view into application/config/bands.php --> 2m are definitive in the array there.
Where is the JS-Array build? Perhaps there's another Option where i (accidently) disabled 2m? In the Frontend it's also marked as active within the band.php-Settings.

But if i really disabled it - it's still a bug, because the default of "2m" are hardcoded, wether it exists nor not.

[image]

int2001, 2022-11-28:
> You didnt by chance disable 2m in the Account -> Band options did you?

i didn't - see update above.

magicbug, 2022-11-28:
Perfect, so I've set the default to "All" hopefully this will resolve it

int2001, 2022-11-28:
Tnx for fast resolving.

PS: This issue could have been a Pullrequest ;) // 73!

phl0, 2022-11-28:
I can reproduce it if I disable 2m in bands. But anyway "All" is the better value to start from with bands being configurable.

[image]