magicbug/Cloudlog #1813: Bug: upload to eQSL shows

philpem, 2022-11-28:
**Describe the bug**

When uploading to eQSL, I have two QSOs which have no eQSL QTH name set. When I look at these QSOs, they have a QTH name set.

My instance is publicly visible at https://logbook.philpem.me.uk

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to eQTH Upload
2. Note that two QSOs are shown which cannot be uploaded as they have no QTH name
3. Open the QSOs and not that they actually do have a QTH name
4. Try to upload the QSOs and receive this error:
 
```
2022-09-06 23:45:15 | SP2EWQ | FT8 |   | 40m | HTTP/1.1 200 OK Content-Type: text/html;charset=UTF-8 Server: Microsoft-IIS/10.0 X-Powered-By: ASP.NET Date: Mon, 28 Nov 2022 14:12:01 GMT Content-Length: 157                Error: Missing eQSL_User

2022-09-06 23:48:53 | LZ2VQ | FT8 |   | 40m | HTTP/1.1 200 OK Content-Type: text/html;charset=UTF-8 Server: Microsoft-IIS/10.0 X-Powered-By: ASP.NET Date: Mon, 28 Nov 2022 14:12:01 GMT Content-Length: 157                Error: Missing eQSL_User
```

**Expected behaviour**
QSO is shown correctly, or not at all

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Linux Mint
 - Browser Firefox
 - Version 107.0 (64-bit)

philpem, 2022-11-28:
I've had a dig in the database and found that I have two entries for the SP2EWQ QSO:
```
MariaDB [cloudlog]> select * FROM TABLE_HRD_CONTACTS_V01 where  COL_CALL = 'SP2EWQ';                                                                                                                                            
+-----------------+-------------+---------+-------------+------------+------------+--------------+---------------+----------+-------------+---------------+----------+-----------+-----------+----------+-------------+----------+------------
------+----------------+-------------+---------+--------------+----------+-----------+-------------+---------------------+---------------------+-------------------+-------------------+-----------------+----------------+----------+--------
-----+----------------+-------------+----------+----------+-------------+---------+---------+---------------------+---------------------+-------------------+-------------------+-----------------+----------------+----------+---------------
+-------------+-------------+----------------+----------------+-------------------+-------------+-----------------+------------+------------+-------------+--------------------+------------+------------+-----------------+--------------+---------------+----------+-----------+---------------+--------------+--------------+--------------------+---------+----------------+---------------+----------------+------------+--------------+--------------+--------------+-----------------
-+--------------+------------------+-------------+------------------+----------------+---------+---------+--------------+--------------+------------+--------------+--------------+---------+---------+--------------+---------+--------------
--+-----------+----------------------+---------+----------------+---------+-------------+---------------------+---------------------+------------+---------+--------------------+--------------------+--------------------+--------------------+--------------------+--------------------+--------------------+--------------------+--------------------+--------------------+--------------------+----------------------+------------------+-------------------+--------------------+-----------------------------+-------------------------------+------------------+------------------+----------------+----------+--------------+-----------+--------------+----------------------------+------------------------------+----------------+---------------------+------------------+---------------------+-------------+--------------+-----------------------+------------------+----------------------+-----------------+-----------------+----------------------+-----------------+--------------------+-----------------------+-------------------+---------------+----------------+----------------------------+------------------------------+--------------+------------------+--------------+------------+--------------+--------------+-------------------+--------------+-------------+-----------+--------------------+----------------+------------+--------------+-----------------+                                                                                 | COL_PRIMARY_KEY | COL_ADDRESS | COL_AGE | COL_A_INDEX | COL_ANT_AZ | COL_ANT_EL | COL_ANT_PATH | COL_ARRL_SECT | COL_BAND | COL_BAND_RX | COL_BIOGRAPHY | COL_CALL | COL_CHECK | COL_CLASS | COL_CNTY | COL_COMMENT | COL_CONT | COL_CONTACTED_OP | COL_CONTEST_ID | COL_COUNTRY | COL_CQZ | COL_DISTANCE | COL_DXCC | COL_EMAIL | COL_EQ_CALL | COL_EQSL_QSLRDATE   | COL_EQSL_QSLSDATE   | COL_EQSL_QSL_RCVD | COL_EQSL_QSL_SENT | COL_EQSL_STATUS | COL_FORCE_INIT | COL_FREQ | COL_FREQ_RX | COL_GRIDSQUARE | COL_HEADING | COL_IOTA | COL_ITUZ | COL_K_INDEX | COL_LAT | COL_LON | COL_LOTW_QSLRDATE   | COL_LOTW_QSLSDATE   | COL_LOTW_QSL_RCVD | COL_LOTW_QSL_SENT | COL_LOTW_STATUS | COL_MAX_BURSTS | COL_MODE | COL_MS_SHOWER | COL_MY_CITY | COL_MY_CNTY | COL_MY_COUNTRY | COL_MY_CQ_ZONE | COL_MY_GRIDSQUARE | COL_MY_IOTA | COL_MY_ITU_ZONE | COL_MY_LAT | COL_MY_LON | COL_MY_NAME | COL_MY_POSTAL_CODE | COL_MY_RIG | COL_MY_SIG | COL_MY_SIG_INFO | COL_MY_STATE | COL_MY_STREET | COL_NAME | COL_NOTES | COL_NR_BURSTS | COL_NR_PINGS | COL_OPERATOR | COL_OWNER_CALLSIGN | COL_PFX | COL_PRECEDENCE | COL_PROP_MODE | COL_PUBLIC_KEY | COL_QSLMSG | COL_QSLRDATE | COL_QSLSDATE | COL_QSL_RCVD | COL_QSL_RCVD_VIA | COL_QSL_SENT | COL_QSL_SENT_VIA | COL_QSL_VIA | COL_QSO_COMPLETE | COL_QSO_RANDOM | COL_QTH | COL_RIG | COL_RST_RCVD | COL_RST_SENT | COL_RX_PWR | COL_SAT_MODE | COL_SAT_NAME | COL_SFI | COL_SIG | COL_SIG_INFO | COL_SRX | COL_SRX_STRIN
G | COL_STATE | COL_STATION_CALLSIGN | COL_STX | COL_STX_STRING | COL_SWL | COL_TEN_TEN | COL_TIME_OFF        | COL_TIME_ON         | COL_TX_PWR | COL_WEB | COL_USER_DEFINED_0 | COL_USER_DEFINED_1 | COL_USER_DEFINED_2 | COL_USER_DEFINED_3
 | COL_USER_DEFINED_4 | COL_USER_DEFINED_5 | COL_USER_DEFINED_6 | COL_USER_DEFINED_7 | COL_USER_DEFINED_8 | COL_USER_DEFINED_9 | COL_CREDIT_GRANTED | COL_CREDIT_SUBMITTED | COL_ADDRESS_INTL | COL_AWARD_GRANTED | COL_AWARD_SUMMITED | COL_C
LUBLOG_QSO_UPLOAD_DATE | COL_CLUBLOG_QSO_UPLOAD_STATUS | COL_COMMENT_INTL | COL_COUNTRY_INTL | COL_SILENT_KEY | COL_SKCC | COL_DARC_DOK | COL_FISTS | COL_FISTS_CC | COL_HRDLOG_QSO_UPLOAD_DATE | COL_HRDLOG_QSO_UPLOAD_STATUS | COL_MY_ANTEN$
[... cut ...] +----------------------+-----------------+-----------------+----------------------+-----------------+--------------------+-----------------------+-------------------+---------------+----------------+----------------------------+------------------------------+--------------+------------------+--------------+------------+--------------+--------------+-------------------+--------------+-------------+-----------+--------------------+----------------+------------+--------------+-----------------+
```

magicbug, 2022-11-28:
OK so 291 doesn't even have a profile id assigned to it so looks that's actually the issue those QSOs either need deleted or the correct ID assigned.station_id

philpem, 2022-11-28:
Whatever's wrong with them also stops them appearing in the search results.

[image]

The link there goes to `javascript:displayQso(292)` -- which is the correct SP2EWQ QSO, not the one that's causing a problem with eQSL.

magicbug, 2022-11-28:
How are you adding FT-x QSOs as it would appear that's causing the problem.

philpem, 2022-11-28:
> How are you adding FT-x QSOs as it would appear that's causing the problem.

Either import from WSJT-X ADIF after-the-fact, or the "wsjtx_cloudlog.py" uploader script../

magicbug, 2022-11-28:
OK Manual ADIF would always set the Station_ID so guess python script needs checked

philpem, 2022-11-28:
Thinking out loud, I may have imported them from eQSL -- but the big issue for me is that I can't see them in the search results. Otherwise I'd just delete them as errors and move on with my life :)

magicbug, 2022-11-28:
Cloudlog won't show anything without Station_ID its big part of the system so i'd suggest giving the records an ID..

philpem, 2022-11-28:
> Cloudlog won't show anything without Station_ID its big part of the system so i'd suggest giving the records an ID..

I would, except I can't open them. They don't appear in the search results when I search for the callsign.

magicbug, 2022-11-28:
Delete directy in the database

```sql
DELETE FROM TABLE_HRD_CONTACTS_V01 WHERE COL_PRIMARY_KEY ='290';
DELETE FROM TABLE_HRD_CONTACTS_V01 WHERE COL_PRIMARY_KEY ='291';
```

phl0, 2022-11-28:
> I would, except I can't open them. They don't appear in the search results when I search for the callsign.

I guess then only direct SQL would make you happy. For the future you might want to check WSJT-X improved. There is a version that has direct Cloudlog API support built-in. No need for external script stuff. 😎✌️

philpem, 2022-11-28:
> Delete directy in the database
> 
> ```sql
> DELETE FROM TABLE_HRD_CONTACTS_V01 WHERE COL_PRIMARY_KEY ='290';
> DELETE FROM TABLE_HRD_CONTACTS_V01 WHERE COL_PRIMARY_KEY ='291';
> ```

That fixed it, thanks! :)

> I guess then only direct SQL would make you happy.

No, I just didn't want to edit the database manually in case I made it inconsistent.

closed 2022-11-28 by magicbug