magicbug/Cloudlog #1816: Date format on previous contacts not the same as anywhere else in the system

PD0RZH, 2022-11-30:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
Date format on previous contacts not the same as anywhere else in the system

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to QSO > Live QSO
2. Click on fill in a callsign which you've worked before
3. See dicsrepancy between date on live QSO page and date format on previous contact
Note: In cloudlog.php the dateformat is set as d/m/y

**Expected behaviour**
Same dateformat throughout the system

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
 - OS: Win 11
 - Browser Edge
 - Version 107.0.1418.62

PD0RZH, 2022-11-30:
Same goes for EQSL upload and download
[image]

phl0, 2022-11-30:
> Same goes for EQSL upload and download [image]

This can be fixed with https://github.com/magicbug/Cloudlog/pull/1817.

closed 2022-12-01 by magicbug