magicbug/Cloudlog #1822: QSL Upload - filetype not allowed (JPG/jpg)

dg9vh, 2022-12-03:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
dev-branch: upload JPG or jpg-file as qsl-card to contact ends in "filetype not allowed"

**Expected behaviour**
JPG or jpg-files should be accepted.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - Linux (Debian)
 - Chrome Version 108.0.5359.71 (Offizieller Build) (64-Bit)

phl0, 2022-12-03:
Strange. Works flawlessly here ....

[image]

dg9vh, 2022-12-03:
hm...
kimhu@DG9VH-LinuxNB:~/Downloads$ file SP6TP.JPG 
SP6TP.JPG: PNG image data, 800 x 523, 8-bit/color RGBA, non-interlaced

dg9vh, 2022-12-03:
OK, there is the problem... it is a PNG in costume of a JPG :-)

phl0, 2022-12-03:
Aaaaahhhh QSL by PA4R QSL tools?

dg9vh, 2022-12-03:
Fixed after rename...
nope... Log4eQSL by https://qrztr.com/ as I can see in the generated email.

dg9vh, 2022-12-03:
Do we still need the issue or should I close it? Problem is not originated by cloudlog, so I think, I would close it :-)

closed 2022-12-03 by dg9vh