magicbug/Cloudlog #1823: Feature Request: Show Frequency plus Band

int2001, 2022-12-05:
When i work S&P, it's very helpful to have a look on the specific Frequency of the last QSO's.
So i made a quick patch to the Views and included COL_FREQ via a dirty Hack to all Logbook-Views.

it's really dirty, so i don't want to do a pullrequest. F.example

```
diff --git a/application/views/qso/index.php b/application/views/qso/index.php
index f849b1c3..e91c92a9 100755
--- a/application/views/qso/index.php
+++ b/application/views/qso/index.php
@@ -550,7 +550,7 @@
                     <?php if($row->COL_SAT_NAME != null) { ?>
                     <td><?php echo $row->COL_SAT_NAME; ?></td>
                     <?php } else { ?>
-                    <td><?php echo $row->COL_BAND; ?></td>
+                   <td title="<?php echo strval($row->COL_FREQ); ?>"><?php echo $row->COL_BAND.'/'.round($row->COL_FREQ/1000,0); ?></td>
                     <?php } ?>
                   </tr>
               <?php $i++; } } ?>
```

I'm not really familiar with the MVC-Pattern in Cloudlog. Perhaps it's easy for the Maintainers to implement this. I think it's really helpful.

phl0, 2022-12-05:
This code has some problems as it fails if COL_FREQ is not set (it is not mandatory). Are we talking about the usual QSO logging screens or the contest logging screen?

int2001, 2022-12-05:
i patched:
```
application/models/Logbook_model.php
application/views/dashboard/index.php
application/views/qso/index.php
application/views/view_log/partial/log_ajax.php
```
So it's now in the Dashboardscreen, Logscreen and the Logbookviewer

phl0, 2022-12-05:
> i patched:
> 
> ```
> application/models/Logbook_model.php
> application/views/dashboard/index.php
> application/views/qso/index.php
> application/views/view_log/partial/log_ajax.php
> ```

So the patch above is incomplete.

int2001, 2022-12-05:
yes. i wrote "f.example" - because i'm not familar with the MVC-Pattern in PHP.
Don't know where it is missing also.

phl0, 2022-12-05:
To be honest: Wouldn't it be more usable if we displayed frequency instead of band instead of both? I mean what difference does it make if you worked me on 7.002 and 7.003?

int2001, 2022-12-05:
Sure. Would be a lot nicer. It was a quick hack for me. Premium-Solution would be a chooser in settings, where you can choose between Band an Freq.

phl0, 2022-12-12:
I have implemented a patch for this. See https://github.com/phl0/Cloudlog/tree/bandFreq. It allows to select Frequency instead of Band in the user selectable columns. If you select Frequency you can also see the frequency of the previous QSOs during logging:

[image]

[image]

Would that suit your needs?

int2001, 2022-12-12:
Great! More than i expected!

AndreasK79, 2022-12-14:
@phl0 PR coming up?

phl0, 2022-12-14:
Cheers: https://github.com/magicbug/Cloudlog/pull/1844

phl0, 2022-12-18:
Merged into the dev branch. @int2001 please test and report. tnx.

int2001, 2022-12-19:
Works like charm.

Tested:
- Dashboard
- Logbook-View
- Creating/Logging QSO on HF and on QO-100 (SAT shows Sat-Name, thats fine)
- Last-QSO-View
- Configpage (switching between Band/Freq)
- Searchpage/View

Anything, which i have forgotten?

vy 73, tnx, DJ7NT

phl0, 2022-12-19:
Looks fine. Tnx. For SAT QSOs it should show the (TX) frequency as tooltip if you hover the SAT name ;-) So if all is done please close here :)

int2001, 2022-12-19:
ah yes - it does. But i have no Sync for SDRConsole, so i always log the QO100-QSOs with 13cm. too lazy :)

int2001, 2022-12-19:
Oh. wait.
there's a bug in "advanced search"

Head is broken and Freq isn't shown.

[image]

AndreasK79, 2022-12-19:
@int2001 pull latest dev and try again. Should be fixed now.

int2001, 2022-12-19:
Hmm. Not really:

[image]

AndreasK79, 2022-12-19:
That explains why it worked ok here. Try another pull. Frequency should work (it works in my dev branch now)

phl0, 2022-12-19:
It shows the band in case frequency is not saved for the QSO. So maybe works as designed here and the QSOs shown are just stored without frequency information?

int2001, 2022-12-19:
it's now perfect!!

phl0, 2022-12-19:
> it's now perfect!!

Then please close this issue ;-) mni tnx

closed 2022-12-19 by AndreasK79