magicbug/Cloudlog #1850: Unable to update DXCC Data

elitegoodguy, 2022-12-21:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
I'm attempting to update DXCC Data

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'Admin > Update Country Files'
2. Click on 'Update DXCC Data'

**Expected behaviour**
I would expect the status to update and show updated with the number of DXCC Entities, Exceptions and prefixes

**Screenshots**
[Screenshot](https://photos.app.goo.gl/kw6gr1byGwUrcHoR8)

**Desktop (please complete the following information):**
 - OS: Alpine Linux
 - Browser: chrome
 - Version [e.g. 22]

**Additional context**
I upped the logging to Level 4 and this is the log that was generated
```
INFO - 2022-12-21 17:27:24 --> Config Class Initialized
INFO - 2022-12-21 17:27:24 --> Hooks Class Initialized
INFO - 2022-12-21 17:27:24 --> UTF-8 Support Enabled
INFO - 2022-12-21 17:27:24 --> Utf8 Class Initialized
INFO - 2022-12-21 17:27:24 --> URI Class Initialized
INFO - 2022-12-21 17:27:24 --> Router Class Initialized
INFO - 2022-12-21 17:27:24 --> Output Class Initialized
INFO - 2022-12-21 17:27:24 --> Security Class Initialized
INFO - 2022-12-21 17:27:24 --> Input Class Initialized
INFO - 2022-12-21 17:27:24 --> Language Class Initialized
INFO - 2022-12-21 17:27:24 --> Loader Class Initialized
INFO - 2022-12-21 17:27:24 --> Config file loaded: /htdocs/application/config/cloudlog.php
INFO - 2022-12-21 17:27:24 --> Config file loaded: /htdocs/application/config/bands.php
INFO - 2022-12-21 17:27:24 --> Config file loaded: /htdocs/application/config/lotw.php
INFO - 2022-12-21 17:27:24 --> Helper loaded: url_helper
INFO - 2022-12-21 17:27:24 --> Helper loaded: security_helper
INFO - 2022-12-21 17:27:24 --> Language file loaded: language/english/general_words_lang.php
INFO - 2022-12-21 17:27:24 --> Database Driver Class Initialized
INFO - 2022-12-21 17:27:24 --> Session: Class initialized using 'files' driver.
DEBUG - 2022-12-21 17:27:24 --> cURL Class Initialized
INFO - 2022-12-21 17:27:24 --> Migrations Class Initialized
INFO - 2022-12-21 17:27:24 --> Language file loaded: language/english/migration_lang.php
INFO - 2022-12-21 17:27:24 --> Database Forge Class Initialized
INFO - 2022-12-21 17:27:24 --> Model "Options_model" initialized
INFO - 2022-12-21 17:27:24 --> Controller Class Initialized
DEBUG - 2022-12-21 17:27:24 --> Migration class already loaded. Second attempt ignored.
INFO - 2022-12-21 17:27:26 --> Config Class Initialized
INFO - 2022-12-21 17:27:26 --> Hooks Class Initialized
INFO - 2022-12-21 17:27:26 --> UTF-8 Support Enabled
INFO - 2022-12-21 17:27:26 --> Utf8 Class Initialized
INFO - 2022-12-21 17:27:26 --> URI Class Initialized
DEBUG - 2022-12-21 17:27:26 --> No URI present. Default controller set.
INFO - 2022-12-21 17:27:26 --> Router Class Initialized
INFO - 2022-12-21 17:27:26 --> Output Class Initialized
INFO - 2022-12-21 17:27:26 --> Security Class Initialized
INFO - 2022-12-21 17:27:26 --> Input Class Initialized
INFO - 2022-12-21 17:27:26 --> Language Class Initialized
INFO - 2022-12-21 17:27:26 --> Loader Class Initialized
INFO - 2022-12-21 17:27:26 --> Config file loaded: /htdocs/application/config/cloudlog.php
INFO - 2022-12-21 17:27:26 --> Config file loaded: /htdocs/application/config/bands.php
INFO - 2022-12-21 17:27:26 --> Config file loaded: /htdocs/application/config/lotw.php
INFO - 2022-12-21 17:27:26 --> Helper loaded: url_helper
INFO - 2022-12-21 17:27:26 --> Helper loaded: security_helper
INFO - 2022-12-21 17:27:26 --> Language file loaded: language/english/general_words_lang.php
INFO - 2022-12-21 17:27:26 --> Database Driver Class Initialized
INFO - 2022-12-21 17:27:26 --> Session: Class initialized using 'files' driver.
DEBUG - 2022-12-21 17:27:26 --> cURL Class Initialized
INFO - 2022-12-21 17:27:26 --> Migrations Class Initialized
INFO - 2022-12-21 17:27:26 --> Language file loaded: language/english/migration_lang.php
INFO - 2022-12-21 17:27:26 --> Database Forge Class Initialized
INFO - 2022-12-21 17:27:26 --> Model "Options_model" initialized
INFO - 2022-12-21 17:27:26 --> Controller Class Initialized
INFO - 2022-12-21 17:27:26 --> Model "Logbook_model" initialized
INFO - 2022-12-21 17:27:26 --> Model "User_Model" initialized
INFO - 2022-12-21 17:27:27 --> Model "DXCC_Entities" initialized
ERROR - 2022-12-21 17:27:27 --> Severity: error --> Exception: Call to undefined function simplexml_load_file() /htdocs/application/controllers/Update.php 41
```

phl0, 2022-12-21:
Is php7.x-xml installed?

elitegoodguy, 2022-12-21:
You are very correct!  I had it installed with PHP8 (Not 8.1 per instructions)  had php8-xml loaded and it wasn't working.  But after installing php8-simplexml it is working now.  It was a problem between chair and keyboard.  Closing this out

elitegoodguy, 2022-12-21:
Closing due to not having all the dependency's installed

closed 2022-12-21 by elitegoodguy