magicbug/Cloudlog #1857: OQRS - requests not shown

dg9vh, 2022-12-22:
**Describe the bug**
After requesting a QSO for QSL, the Request is not shown in the requests-view.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to /oqrs as not logged in user
2. Request QSL for QSO
[image]
3. Open requests-pag
[image]
5. See error

The QSO is correctly marked in the database-table:
[image]

**Expected behaviour**
Find the QSO as requested in the requests-view.

AndreasK79, 2022-12-22:
Since it's a match, you'll fknd it in qslprint.

dg9vh, 2022-12-22:
But it would be printed as bureau-card, there is no thing saying it is direct:
[image]

AndreasK79, 2022-12-22:
You are right. Another issue to look into. I bet there are more, but the only way to get feedback here, was to release it.

AndreasK79, 2022-12-22:
Bureau or direct is never set on the QSO, only on the OQRS I see. Thanks for reporting. Will hopefully fix it tomorrow. Just pushed several SAT fixes to dev.

dg9vh, 2022-12-22:
> You are right. Another issue to look into. I bet there are more, but the only way to get feedback here, was to release it.

Yes, but it was very short in /dev so I had no chance to report this earlier :-)
I am running dev-branch in production and would find this issues there vy soon :-)

AndreasK79, 2022-12-22:
Yes, the release was a bit quick, but we'll fix the stuff :) Just keep testing and reporting. More to find!

magicbug, 2022-12-22:
> , but the only way to get feedback here, was to release it.

Exactly why I did it

AndreasK79, 2022-12-23:
@dg9vh pushed fix to DEV now. Turns out the QSO class which logbookadvanced uses, is case sensitive.

Besides, I think Print Requested QSLs should be updated to show col_qsl_sent_via as well.

dg9vh, 2022-12-23:
Ok, will test it tomorrow, if time available.

dg9vh, 2022-12-24:
Requested new QSL as direct, but it is now shown as requested direct in QSO-View:
[image]

Working :-)

> Besides, I think Print Requested QSLs should be updated to show col_qsl_sent_via as well.

Would you fix this within this issue or should I file another one?

AndreasK79, 2022-12-24:
No need for another issue. Just need some time:)

AndreasK79, 2022-12-26:
Updated dev now so that QSLPrint shows sent method.

closed 2022-12-26 by AndreasK79