magicbug/Cloudlog #1863: HTTP500 during ADIF export

magicbug, 2022-12-22:
### Discussed in https://github.com/magicbug/Cloudlog/discussions/1862

<div type='discussions-op-text'>

<sup>Originally posted by **HakanKoseoglu** December 22, 2022</sup>
Possible regression. I am not sure if I should raise an issue. Can anyone else replicate this?

Code at: commit 8b9d2598ba248e9cff60ba8b26e1f25f43d4833a (HEAD -> master, tag: 2.3, origin/master, origin/HEAD)

Edit: ran it on developer mode, additional log:

_An uncaught Exception was encountered
Type: Error

Message: Call to a member function format() on bool

Filename: /var/www/html/application/models/Adif_data.php

Line Number: 87_

Open ADIF Import/Export, select Export.

Selected logbook, did not click on any other options. Fails with HTTP500 as below, which isn't very helpful. 

172.20.0.1 - - [22/Dec/2022:22:29:24 +0000] "POST /index.php/adif/export_custom HTTP/1.1" 500 295 "https://logbook.koseoglu.org/index.php/adif" "Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/108.0.0.0 Safari/537.36"

If I enter start/end dates, this executes w/o a problem.

[image]
</div>

magicbug, 2022-12-22:
Hmm just testing this on my live and dev

- Error if I don't select a Location
- ADIF generates fine if dates or filled in or blank.

What PHP version?

pa8s, 2022-12-23:
I can reproduce your error on my system. 
code at commit 5d2ca034
PHP 8.1.13

closed 2022-12-24 by magicbug