magicbug/Cloudlog #1865: Result 'undefined' when plotting Distances Worked

pa8s, 2022-12-23:
**Describe the bug**
While trying to plot the Distances Worked, on all bands the result is a red 'undefined' error box.

**To Reproduce**
Steps to reproduce the behaviour:

1. Analytics -> Distances Worked -> any band -> Plot

**Expected behaviour**
The usual Distances Worked diagram.

**Screenshots**
[image]

**Logfile**
Below a part of the logfile of one explicit distance(?) value. Similar blocks of errors (line 180 through 186) are repeated thousands of times per query.
```
ERROR - 2022-12-23 12:37:20 --> Severity: 8192 --> Implicit conversion from float 112.26 to int loses precision /mnt/www/application/models/Distances_model.php 180
ERROR - 2022-12-23 12:37:20 --> Severity: 8192 --> Implicit conversion from float 112.26 to int loses precision /mnt/www/application/models/Distances_model.php 181
ERROR - 2022-12-23 12:37:20 --> Severity: 8192 --> Implicit conversion from float 112.26 to int loses precision /mnt/www/application/models/Distances_model.php 182
ERROR - 2022-12-23 12:37:20 --> Severity: 8192 --> Implicit conversion from float 112.26 to int loses precision /mnt/www/application/models/Distances_model.php 183
ERROR - 2022-12-23 12:37:20 --> Severity: 8192 --> Implicit conversion from float 112.26 to int loses precision /mnt/www/application/models/Distances_model.php 185
ERROR - 2022-12-23 12:37:20 --> Severity: 8192 --> Implicit conversion from float 112.26 to int loses precision /mnt/www/application/models/Distances_model.php 186

```

**Server**
OS: FreeBSD 13.1
Server: nginx/1.22.1
PHP: 8.1.13
Git branch: dev
Git commit: 5d2ca034
Git Tag: 2.1-125-g5d2ca034

AndreasK79, 2022-12-24:
This seems to be PHP 8.1 related. I'm getting this on my PHP8.1 test machine, but my PHP7.4 install works ok. I'll see if I can figure out what is going on. Thanks for reporting.

AndreasK79, 2022-12-24:
@pa8s Pushed a fix to the dev branch now.

pa8s, 2022-12-24:
Plotting works again, thanks. :)

closed 2022-12-24 by magicbug