magicbug/Cloudlog #1870: LoTW upload broken?

pa8s, 2022-12-24:
**Describe the bug**
Uploading/synchronisation with LoTW results in error on webpage and log.

**To Reproduce**
Steps to reproduce the behaviour:

1. Click on 'Your Call'
2. Click Logbook of the World 
3. Click Manual Sync

**Expected behaviour**
Synchronisation with LoTW

**Screenshots**
[image]

**Logfile**
```
ERROR - 2022-12-24 14:29:01 --> Severity: 8192 --> Function openssl_free_key() is deprecated /mnt/www/application/controllers/Lotw.php 913
INFO - 2022-12-24 14:29:01 --> File loaded: /mnt/www/application/views/lotw_views/adif_views/adif_export.php
ERROR - 2022-12-24 14:29:01 --> Severity: error --> Exception: Call to undefined function gzencode() /mnt/www/application/controllers/Lotw.php 277
```

**Server**
OS: FreeBSD 13.1
Server: nginx/1.22.1
PHP: 8.1.13
Git branch: dev
Git commit: 8346b69e
Git Tag: 2.1-133-g8346b69e

magicbug, 2022-12-24:
I'd suggest you check if gzencode is available as its a PHP function

pa8s, 2022-12-24:
When upgrading from php 8.0 to 8.1, I cleaned up some unused extensions, accidentally in the process I also removed zlib that takes care of gzencode, among other things. Small mistake on my part, so this issue can be ignored.

closed 2022-12-24 by pa8s