magicbug/Cloudlog #1875: [PHP 8.1] Error when opening widgets/qsos

pa8s, 2022-12-27:
**Describe the bug**
When opening https://cloudlog-url/index.php/widgets/qsos an 'uncaught exception was encountered'.
The problem occurs regardless of whether one is logged in or not.

**To Reproduce**
Open URL https://cloudlog-url/index.php/widgets/qsos 

**Expected behaviour**
A list of recent QSOs is shown.

**Screenshots**
[image]

**Logfile**
```
ERROR - 2022-12-27 23:09:24 --> Severity: error --> Exception: Too few arguments to function Widgets::qsos(), 0 passed in /mnt/www/system/core/CodeIgniter.php on line 485 and exactly 1 expected /mnt/www/application/controllers/Widgets.php 18
```

**Server**
OS: FreeBSD 13.1
Server: nginx/1.22.1
PHP: 8.1.13
Codeigniter: 3.2.0-dev
Git branch: dev
Git commit:  54c31a6b
Git Tag: 2.1-134-g54c31a6b

**Additional context**
The URL works fine on Cloudlog 1.7(?) and PHP 7.4.

magicbug, 2022-12-28:
If I remember right you need to add the public slug to the end of the URL in V2 but error could be better

closed 2022-12-28 by magicbug

pa8s, 2022-12-28:
Thanks, I had to search a bit for what a 'public slug' is but found it. 

For those also looking for it:  `<your account> -> Station Logbooks -> Edit -> Public Slug -> <slugname> -> Save `
Now your list of last QSOs can be found at https://cloudlog-url/index.php/widgets/qsos/slugname

commit 6f597ff7 (2022-12-28): [Widgets] /QSO/{public_slug} Error Message

Fixes #1875 providing a more useful error message