magicbug/Cloudlog #1876: OQRS: Only one QSO found even if there ere more QSOs in log

m0urs, 2022-12-28:
If you search for a call sign in the OQRS page for which there are more than one QSO in the log, only one QSO per band is found:

[image]

[image]

I would have expected that all QSOs on all bands are found.

AndreasK79, 2022-12-28:
No, it should not. Only the distinct should show here.
What if you click on Request QSL?

AndreasK79, 2022-12-28:
But I see that the text is wrong. Should list band 2, and 6 QSOs in log.

m0urs, 2022-12-28:
I get one line per band. so I cannot request a QSL for more than one QSO on the same band?

AndreasK79, 2022-12-28:
I see. Need to check out the code then. Table should show distinct, but the QSO list should list them all. Thanks for the report.

AndreasK79, 2022-12-30:
I found the issue. It is fixed in #1883.

closed 2022-12-30 by AndreasK79

m0urs, 2022-12-30:
Can confirm that this solves the issue.