magicbug/Cloudlog #1878: Unable to delete QSL Card

elitegoodguy, 2022-12-28:
This was happening before the 2.3 upgrade however I didn't get a chance to really look into the logs until now.  

I uploaded some QSL cards however I accidentally added them to the wrong QSO.  

I go to delete and nothing happens.  I have also tried to delete via QSO > View QSL > Delete and the same.

I upped the logging level of my CloudLog install and I'm getting the error below everytime I click the delete button.  

`ERROR - 2022-12-28 20:04:42 --> Severity: error --> Exception: Call to a member function row() on null /var/www/html/application/controllers/Qsl.php 50`

phl0, 2022-12-28:
Confirmed there is a bug in the getFilename function.

phl0, 2022-12-28:
And patch is ready: https://github.com/magicbug/Cloudlog/pull/1879.

elitegoodguy, 2022-12-28:
Wow that was fast.  Took your patch and applied it and is working!  Thank you!  

Looking forward to contributing to Cloudlog myself.

phl0, 2022-12-28:
> Wow that was fast. Took your patch and applied it and is working! Thank you!

Your welcome. Thank you for testing.

> Looking forward to contributing to Cloudlog myself.

Always welcome :)

elitegoodguy, 2022-12-28:
Closing due to patch was created to fix this issue

closed 2022-12-28 by elitegoodguy