magicbug/Cloudlog #1897: Implicit conversion from float to int loses precision

pa8s, 2023-01-02:
**Describe the bug**
When manually entering a QSO, while tabbing through the input fields into the 'Gridsquare' field a PHP error occurred.

**To Reproduce**
Steps to reproduce the behaviour:
1. QSO
2. Post QSO
3. Enter sensible values
4. Tab until in 'Gridsquare'  field.
5. Error is shown. 

**Expected behaviour**
No error. 

**Screenshots**
[image]

**Logfile**
```
ERROR - 2023-01-02 15:35:05 --> Severity: 8192 --> Implicit conversion from float 210.65327391603532 to int loses precision /mnt/www/application/libraries/Qra.php 168
```

**Server**
OS: FreeBSD 13.1
Server: nginx/1.22.1
PHP: 8.1.13
Git branch: dev
Git commit: 52a37802
Git Tag: 2.1-179-g52a37802

**Additional context**
Is this similar to https://github.com/magicbug/Cloudlog/issues/1865 ??
Perhaps this occurs in more places than what I have found so far.

AndreasK79, 2023-01-02:
@pa8s thanks again for reporting. I've pushed a fix to the dev branch.

closed 2023-01-02 by AndreasK79

pa8s, 2023-01-02:
Another quick fix. However, I just noticed that the tooltip near the gridsquare field has some, to me, strange text. How can you have a QSO to multiple gridsquares? Is this also worth writing an issue?

[image]

phl0, 2023-01-03:
> How can you have a QSO to multiple gridsquares? Is this also worth writing an issue?

This is intended. For ARRL VUCC awards it is valid to count both squares if you are located at the grid line. Even for grids is possible if you are at the intersection of the two lines of four squares.

pa8s, 2023-01-03:
> > How can you have a QSO to multiple gridsquares? Is this also worth writing an issue?
> 
> This is intended. For ARRL VUCC awards it is valid to count both squares if you are located at the grid line. Even for grids is possible if you are at the intersection of the two lines of four squares.

Okay, still sounds a bit strange to me but I get the intention.

phl0, 2023-01-03:
> Okay, still sounds a bit strange to me but I get the intention.

It is very common among SAT operators ;-)

magicbug, 2023-01-03:
Definitely isn't a bug on VHF+ being on gridlines is common place :)