magicbug/Cloudlog #1899: Feature request: OS detection in update_cloudlog.sh

pa8s, 2023-01-03:
**Describe feature**
Currently, the script sets the ownership to root:www-data which is specific to Linux. I have Cloudlog running on FreeBSD and there the default user/group for a web page is www:www. Instead of manually adjusting the ownership after each update, it would be nice if the update script would detect the OS and adjust the ownership accordingly. 

**Suggestion**
Expand the `update_cloudlog.sh` script with something like:

```
# The user and group that own the CLOUDLOG_SUBDIR directories. Passed to 'chown' as-is.
# Based on detected operating system.
if [[ "$OSTYPE" == "FreeBSD" ]]; then
       DIR_OWNERSHIP="www:www"
else
       DIR_OWNERSHIP="root:www-data"
fi
```
I don't know what the default ownership for web pages on Windows or OSX are but that could also be extended here.

magicbug, 2023-01-03:
While nice think there has to be a point where it's beyond expected scope of something.

pa8s, 2023-01-03:
Fine, it was nothing more than an idea by the way. :) I don't know if there are other non-Linux Cloudlog users.

closed 2023-01-06 by pa8s