magicbug/Cloudlog #1904: wants to keep running the install

anwoke8204, 2023-01-07:
Hi, I uploaded this to my webhost, ran through the install, but it keeps wanting to run the install, even though I did the install and it said the install succeeded and then told me to delete the install folder.  how can I get past this?

AndreasK79, 2023-01-07:
Hi. If the config.php file is not found, it will redirect you to run install again.
It's impossible to say what might be wrong without knowing more about your setup.
Config.php should exist in application/config/ after install is done. If not, some permissions might not be set correctly.

magicbug, 2023-01-07:
I'm guessing permissions issue on application/config folder so the installer isn't creating the config files thus install loop

closed 2023-01-25 by magicbug