magicbug/Cloudlog #1907: Station Lookup not working

do4bmw, 2023-01-09:
Hy

i have a problem that my QRZ.com Lookup does not work.

If i enter a callsign in the LiveQSO Window , in the past it makes a fetch of the name , locator , etc

but since update to 2.x it does not work.

Auto look up and image fetch option are enabled

the QSO upload to QRZ.com works fine. and i have a QRZ.com subscription

maybe the 2FA on QRZ.com is a problem ? or something else ?

[image]
[image]
[image]

do4bmw, 2023-01-09:
[image]

do4bmw, 2023-01-09:
here as example with the clubstation callsign DL0GH

there are no infos , or picture ... but if you look at [qrz.com/db/dl0gh](url) there is

phl0, 2023-01-09:
No issues known in 2.0. Anything useful in the log or browser console?

do4bmw, 2023-01-09:
what logfile ? apache ?

here a screenshot from the browser console
[image]

magicbug, 2023-01-09:
I'd make sure that /application/config/config.php is correctly setup for QRZ.

do4bmw, 2023-01-09:
> I'd make sure that /application/config/config.php is correctly setup for QRZ.
> 
> You could also check application/logs to see if anything is getting reported
> 
> But certainly not aware of any bugs in this code.

that was the problem ! many thanks , problem solved !

i had 

$config['callbook'] = "hamqth"; // Options are hamqth or qrz

and now changed to 

$config['callbook'] = "qrz"; // Options are hamqth or qrz

because i dont use hamqth ... my failure

many thanks for help !

closed 2023-01-09 by do4bmw