magicbug/Cloudlog #1923: Feature Request: Use "Sent Method" for "Mark requested QSLs as sent"

m0urs, 2023-01-12:
If you mark a QSO for "QSL requested", you can also set the method, e.g. "Electronic":

[image]

If you later, in the "Print requested QSLs" view, press the button "Mark requested QSLs as sent", the QSO will always be marked as "QSL sent via bureau":

[image]

It would be very helpful, if the "Sent method" is not changed to bureau, but would be left with its original value.

If there is no sent method choosen before, then "Bureau" as default would be ok.

What do you think?

AndreasK79, 2023-05-03:
@m0urs I agree. Done and pushed to dev.

m0urs, 2023-05-03:
Thanks very much. This does work for me and is very helpful!

closed 2023-05-03 by m0urs