magicbug/Cloudlog #1929: Extern modified field is not correct updated in the database

HB9WDF, 2023-01-15:
Last November, I logged >80 QSO with the Sat "Greencube".
After the announcement  of Change from "Greencube" to IO-177, I exported all the "Greencube" QSOs in a ADIF File.
I changed the satellit name with the "ADIF Master" app and reimported all the IO-117 QSO's to Cloudlog.
After the import, I uploaded all the QSOs to LotW and eQSL.
There, I had no Issue.
I have deleted all the QSO with the sat name "Greencube".

All looks perfect...

Today, I saw, that >80 QSOs are not uploaded to QRZ Logbook. So I uploaded all the QSO's to QRZ Logbook.

Result: "0 QSOs are now uploaded to QRZ.com"
 In the error message, I got this information: "STATUS=FAIL&REASON=Unable to add QSO to database: duplicate&EXTENDED= Call: AB1CDE Band: 70cm Mode: PKT Time: 2022-11-22 20:17:58" (The call AB1CDE is not the true callsign)

Workaround:
Open the qso in the log with "edit QSO" and press "save changes". 

With this workaround, the upload is with success. I think, there is an Issue with the import of the qso. Interresting fact: LotW and eQSL accepted the QSO, QRZ.com not.

magicbug, 2023-01-16:
I'm going to guess this is an QRZ only issue where its not looking at the satellite name just the band and mode which would make it a duplicate.

You certainly didn't need to export and change the name, there's code provided inside Cloudlog todo name change to make life easier for everyone.

phl0, 2023-01-24:
I'd say the only option is to delete the "Greencube" QSOs from qrz and re-upload.