magicbug/Cloudlog #1944: No logbooks were found

teabreakninja, 2023-01-21:
**Describe the bug**
A new logbook doesn't seem to be associated with the user.

**To Reproduce**
Fresh Install of Cloudlog from Git, populate DXCC, create Station info and location. Create new log. Import ADIF backup.
Logbook not associated with user. 

**Expected behaviour**
Backup logs imported into new logbook.

**Desktop (please complete the following information):**
Linux Server, Docker using https://hub.docker.com/r/mattrayner/lamp

**Additional context**
No entry in station_logbooks_relationship table, all imported log entries visible in TABLE_HRD_CONTACTS_V01.

teabreakninja, 2023-01-21:
ok, seems it's missing an entry in station_logbooks_relationship

magicbug, 2023-01-22:
Not a bug, read the error, to go Station Logbooks create one, then link it to any of the Station Locations.

closed 2023-01-22 by magicbug

teabreakninja, 2023-01-22:
No Link option is/was availble

[image]