magicbug/Cloudlog #1945: [PHP 8.1] mb_substr(): Passing null to parameter #2 ($start) of type int is deprecated

pa8s, 2023-01-21:
**Describe the bug**
While importing an ADIF file, mb_substr() deprecated error shown.

**To Reproduce**
Steps to reproduce the behaviour:
1. <Your Call>  -> ADIF Import/Export -> ADIF Import -> <select local file> -> <select station location> -> Mark LoTW -> Upload 
2. See error in screenshot.

**Expected behaviour**
No error.

**Logfile**
```
ERROR - 2023-01-21 21:16:17 --> Severity: 8192 --> mb_substr(): Passing null to parameter #2 ($start) of type int is deprecated /mnt/www/application/libraries/Adif_parser.php 118
```

**Server**
OS: FreeBSD 13.1
Server: nginx/1.22.1
PHP: 8.1.13
Codeigniter: 3.2.0-dev
Git branch: dev
Git commit: b03870e1
Git Tag:  2.1-239-gb03870e1

AndreasK79, 2023-01-21:
Can't reproduce that here. Any way you were willing to share the ADIF-file so I could test?

pa8s, 2023-01-22:
Problematic ADIF record added to issue #1946

AndreasK79, 2023-01-22:
Fix pushed to dev. Variable was never set if EOH didn't exist in the file. And as a result, null was passed to mb_substr.

pa8s, 2023-01-22:
Checked with 'problematic' ADIF file. I had split a large ADIF file into smaller files hence some imports had no header present.

closed 2023-01-22 by pa8s

AndreasK79, 2023-01-22:
And that uncovered a bug that needed fixing :)

pa8s, 2023-01-22:
Apparently, the transition from PHP 7 to 8 is giving you guys a lot of work with all those deprecated functions. Exploratory testing is in my nature, so discovering those bugs is to be expected. :)

AndreasK79, 2023-01-22:
Nothing is better than getting people to test. Most of the bugs are easy to fix. I swapped to PHP8.1 myself on my dev machine, so I've fixed a few that suddenly appeared.