magicbug/Cloudlog #1946: [PHP 8.1] preg_match(): Passing null to parameter #2 ($subject) of type string is deprecated

pa8s, 2023-01-21:
**Describe the bug**
While importing an ADIF file, mb_substr() deprecated error shown.

**To Reproduce**
Steps to reproduce the behaviour:
1. <Your Call>  -> ADIF Import/Export -> ADIF Import -> <select local file> -> <select station location> -> Mark LoTW -> Upload 
2. See error in screenshot.

**Expected behaviour**
No error.

**Logfile**
```
ERROR - 2023-01-21 21:25:41 --> Severity: 8192 --> preg_match(): Passing null to parameter #2 ($subject) of type string is deprecated /mnt/www/application/models/Logbook_model.php 3090
```

**Server**
OS: FreeBSD 13.1
Server: nginx/1.22.1
PHP: 8.1.13
Codeigniter: 3.2.0-dev
Git branch: dev
Git commit: b03870e1
Git Tag:  2.1-239-gb03870e1

AndreasK79, 2023-01-21:
Is this related to #1945? Is it the same ADIF-file? I've not seen this happen, so I'm very curious about the ADIF-file you tried to import.

pa8s, 2023-01-21:
Yes, related to #1945. Happened during importing of several small ADIF files. Just have to see which one caused this.

pa8s, 2023-01-22:
Narrowed down to one ADIF record causing both preg_match() and mb_substr() error (as reported in issue #1945). The QSO below is already present in my logbook when importing the ADIF file.

In the Cloudlog logfile I just saw additional errors being logged:

```
DEBUG - 2023-01-22 12:06:41 --> cURL Class Initialized
ERROR - 2023-01-22 12:06:41 --> Severity: 8192 --> mb_substr(): Passing null to parameter #2 ($start) of type int is deprecated /mnt/www/application/libraries/Adif_parser.php 118
DEBUG - 2023-01-22 12:06:41 --> Frequency class already loaded. Second attempt ignored.
ERROR - 2023-01-22 12:06:41 --> Severity: Warning --> Undefined array key 1 /mnt/www/application/models/Logbook_model.php 3066
ERROR - 2023-01-22 12:06:41 --> Severity: Warning --> Trying to access array offset on value of type null /mnt/www/application/models/Logbook_model.php 3066
ERROR - 2023-01-22 12:06:41 --> Severity: 8192 --> preg_match(): Passing null to parameter #2 ($subject) of type string is deprecated /mnt/www/application/models/Logbook_model.php 3090
```

```
<APP_LoTW_OWNCALL:4>PA8S
<STATION_CALLSIGN:4>PA8S
<MY_DXCC:3>263
<MY_COUNTRY:11>NETHERLANDS
<APP_LoTW_MY_DXCC_ENTITY_STATUS:7>Current
<MY_GRIDSQUARE:6>JO32BL
<MY_CQ_ZONE:2>14
<MY_ITU_ZONE:2>27
<CALL:9>GQ0ACQ/70
<BAND:3>30M
<FREQ:8>10.13784
<MODE:3>FT8
<APP_LoTW_MODEGROUP:4>DATA
<QSO_DATE:8>20220608
<APP_LoTW_RXQSO:19>2022-06-08 20:11:16 // QSO record inserted/modified at LoTW
<TIME_ON:6>190800
<APP_LoTW_QSO_TIMESTAMP:20>2022-06-08T19:08:00Z // QSO Date & Time; ISO-8601
<QSL_RCVD:1>Y
<QSLRDATE:8>20220628
<APP_LoTW_RXQSL:19>2022-06-28 20:02:51 // QSL record matched/modified at LoTW
<DXCC:3>223
<COUNTRY:7>ENGLAND
<APP_LoTW_DXCC_ENTITY_STATUS:7>Current
<PFX:3>GQ0
<APP_LoTW_2xQSL:1>Y
<GRIDSQUARE:6>IO80SN
<CQZ:2>14
<ITUZ:2>27
<eor>
```

AndreasK79, 2023-01-22:
Thanks. I can reproduce the error now. Will look into it.

AndreasK79, 2023-01-22:
I've pushed a fix for one of the issues to the dev branch. The DXCC identification had trouble with /70 in the callsign. I've adjusted it now, so that it is recognized as England.

pa8s, 2023-01-22:
Checked, also a /70 QSO originating from Scotland is handled as expected. :)

closed 2023-01-22 by AndreasK79