magicbug/Cloudlog #1956: Feature request: If only one location set in Cloudlog, set it by default for ADIF import/export

ofadam, 2023-01-26:
For users with only one location set in Cloudlog, it would be handy to show that one by default on the ADIF import/export page instead of forcing them to use the pull-down menu to select it. A minor quibble, but one that would save a click with each import/export!

[image]

magicbug, 2023-01-26:
I think default should be the "active" location of the user even if there's more than one.

ofadam, 2023-01-26:
That makes a lot more sense – will adjust issue title to reflect.

phl0, 2023-01-27:
Proposed a patch for this in https://github.com/magicbug/Cloudlog/pull/1958.

magicbug, 2023-01-27:
@phl0 your on it faster than me these days! merged into dev branch will make it into master next week

closed 2023-01-27 by magicbug

phl0, 2023-01-27:
Oh sorry for that. I saw the issue last night but as it wasn't solved this morning I thought I'd contribute some code :)