magicbug/Cloudlog #1957: Feature request: Match style of advanced logbook with overview logbook

ofadam, 2023-01-26:
The new advanced logbook view is an incredibly powerful page for filtering QSOs, but the overall look borrows heavily from an average spreadsheet with a very large grid.

On the classic overview logbook view, the results have a bit more polish with striped rows and no vertical rules at all.

Obviously, this may come down to personal style preference – but there may be benefits toward standardizing style between the two?

magicbug, 2023-01-26:
@ofadam Fear not, its on my radar as well, needs a bit more polish to fit in with the rest of the system

ofadam, 2023-01-26:
No fear here! Thanks for all you do.

magicbug, 2023-01-26:
It bugged me the other day, got to get the small things right :)

AndreasK79, 2023-04-30:
The main reason not adding this in the first place, was that marking lines would not show correctly.
I've fixed striped table and marking now. I have pushed this to the dev branch.

@magicbug didn't bug you enough to fix? ;)

closed 2023-04-30 by AndreasK79