magicbug/Cloudlog #1971: R2 prefix with /P is seen as European Russia

pa8s, 2023-01-31:
**Describe the bug**
R2 prefix with /P is seen as European Russia (DXCC CQ Zone 16,17,21) while this should be Kaliningrad (CQ Zone 15)

**Server**
OS: FreeBSD 13.1
Server: nginx/1.22.1
PHP: 8.1.13
Codeigniter: 3.2.0-dev
Git branch: [dev](https://github.com/magicbug/Cloudlog/tree/dev)
Git commit: [5e5136fa](https://github.com/magicbug/Cloudlog/commit/5e5136fa4432911b5b6e1fd56d2add077454e89e)
Git Tag: [2.3.1-56-g5e5136fa4](https://github.com/magicbug/Cloudlog/releases/tag/2.3.1)

AndreasK79, 2023-01-31:
Seems the algorithm needs more tweaking. I'll debug and see what happens. Thanks for reporting.

AndreasK79, 2023-02-01:
This one is harder to fix without breaking identification of other DXCC's.

I wrote a call tester (being tired of testing my list manually), so that it's easier to identify cases that should be identified.

We will never get to 100%, but it can probably be improved upon.

magicbug, 2023-02-01:
Even so, /P should be able to not affect the main dxcc lookup as we didn't have these issues before hand although it largely ignored anything after the /

AndreasK79, 2023-02-01:
Ignoring everything after / is not the solution either.

AndreasK79, 2023-02-01:
Added #1977. It fixes issue with Kaliningrad and other issues I found. 

It includes the calltester as well.

AndreasK79, 2023-02-05:
Merged into dev. Closing.

closed 2023-02-05 by AndreasK79