magicbug/Cloudlog #1981: Order by date not work correctly

Stiles96, 2023-02-05:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
In the contest log are displayed old logs from other contest and the time shows at first the old contest and then the current and the newest first.

**To Reproduce**
Steps to reproduce the behaviour:
Show Screeenshot. Make some contest logging.

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Win 10 22H2
 - Browser Firefox
 - Version latest

AndreasK79, 2023-04-30:
I tried to fix this, but couldn't get it to work the way I wanted. I got it to order by date/time correctly when loading the contest session, but when adding a new QSO, it would always add the line out of sight at the bottom. 

Unfortunately, I don't see any way to fix this.

AndreasK79, 2023-05-03:
I tried my fix again, and now it seems to work. I have pushed a fix to dev. Must have been a cached file or something else strange.

closed 2023-05-03 by AndreasK79