magicbug/Cloudlog #1984: Not able to log non-existing WWFF reference

m0urs, 2023-02-05:
I had the problem that I needed to log a very new WWFF reference which had been existing on the WWFF website but seems not to be included in the CSV file (even I updated it manually).

Because of that, I was not able to enter this reference as the field will be wiped out if there is no match found.

I would like to be able to enter a reference, even if it is not found in the CSV file (for sure there is no no update of QTH and locator in this case, but at least I can log the reference number).

phl0, 2023-02-06:
There is this in the DOK field:

[image]

So just need to find the difference here ...

phl0, 2023-02-06:
Solution proposed in https://github.com/magicbug/Cloudlog/pull/1985.
Looks like this then:

[image]

m0urs, 2023-02-06:
Thanks a lot, this does work for me. I created a similar pull request also for POTA and SOTA references (https://github.com/magicbug/Cloudlog/pull/1987)

closed 2023-02-07 by magicbug