magicbug/Cloudlog #1996: [Feature Request] Autoupdate of Locator and QTH also for "Post QSO"

m0urs, 2023-02-09:
Currently the auto update of Locator and QTH field (for WWFF, SOTA, POTA does only work if you are logging a "Live QSO". It would be great if this would work also for "Post QSO" logging.

I thought, I already posted this request in the past, but could not find it anywhere. So sorry, if this is a duplicate

AndreasK79, 2023-04-30:
I just tested this, and it works both in live and post QSO. Are you sure you have it set to on in user settings?

m0urs, 2023-04-30:
Sorry, yes, you are right for the "Post QSO" menu item. I did not describe the issue correctly. What I mean is, that, if you have already added a QSO and then afterwards edit it, the auto update feature is not available. Sometimes I add the QSO before I am exactly sure about the SOTA/WWFF reference number and want to add it later. In this case it would be great if I do not have to enter QSO and locator manually.

By the way: Is it correct, that the auto update feature is not available for POTA references, only for SOTA and WWFF?

AndreasK79, 2023-04-30:
That explains is a lot clearer yes. Since the feature exists, it should probably not be too hard to add in edit mode.

I could only find references to it working for SOTA and WWFF. It all depends on POTA having an API to query for info. I'm not aware of such an API.

phl0, 2023-04-30:
That true. If you can deliver an info about an API endpoint that provides details like park name, grid etc. we can integrate that. I did some research when I implemented this and there wasn't...

m0urs, 2023-04-30:
Do we need an API or would a CSV file which includes reference number, reference description and locator? You can download that from https://pota.app/all_parks_ext.csv

There is also an API (https://api.pota.app) but I did not find any further description for it so far.

We can get also all information directly via that URL https://pota.app/#/park/<park reference> but the result is HTML and we need to locate the needed information somewhere in the HTML code.

phl0, 2023-05-01:
I would recommend using an API that you can query foe a specific park and gives back some JSON with the park details. This is how it works for WWFF and SOTA. 

I know there is HTML but that is a lot more to parse and probably very painful to scrape the information from. Also very error-prone to my experience.

m0urs, 2023-05-01:
I asked the POTA team if there is a documentation for their API. Otherwise: If we use the CSV file I mentioned above, which already includes all necessary information like description and locator, would this be an alternative?

m0urs, 2023-05-01:
Ok, got the necessary information. To get all the information of a POTA reference we can use the following API call

https://api.pota.app/park/DA-0001 (where DA-0001 is the POTA reference)

 which gives us an JSON object which has, besides other, the following information:
 
 "name": "Berchtesgaden"
 "grid6": "JN67ln"
 
 Do we need a separate topic to track that feature request or can we handle that with this topic?

phl0, 2023-05-01:
I think thats enough. I will have a look.

phl0, 2023-05-01:
@m0urs Please test the code in https://github.com/magicbug/Cloudlog/pull/2104.

m0urs, 2023-05-02:
For me it seems to work great. Thanks a lot for adding this!

closed 2023-07-08 by magicbug

m0urs, 2023-07-08:
@magicbug Unfortunately, there were two issues mixed in this feature request. The POTA functionality is working now, but the original feature request "Autoupdate of Locator and QTH also while editing an existing QSO" is not. At least it does not work for me?