magicbug/Cloudlog #2001: ADIF export on Logbook Advanced does not reflect the view created

PD0RZH, 2023-02-12:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
When in Logbook Advanced  and using the ADIF export the view which has been set (for example date/time descending) is not reflected in the actual ADIF export file thus creating errors in the automated log submission form for contests. This automated form expects the QSO's in the right date/time order.

**To Reproduce**
Steps to reproduce the behaviour:
1 Log a few QSO's
2 Go to Logbook advanced
3 Sort on date time (ascending or descending) and take a not which one you have chosen
4 Hit the export to ADIF
5 Repeat 2-4 but with descending or ascending the other way around in step 3
6 check the ADIF file

**Expected behaviour**
The ADIF file should reflect the sorting of the Logbook Advanced view

AndreasK79, 2023-05-01:
The sort order for the table was never sent in. I added so that parameter is sent in when exporting to ADIF.
I did not add support for all columns.

These are the ones that will change sort order:

```
switch($sortorder[0]) {
	case 1: $order = 'ORDER BY qsos.COL_TIME_ON ' . $sortorder[1]; break;
	case 2: $order = 'ORDER BY station_profile.station_callsign ' . $sortorder[1]; break;
	case 3: $order = 'ORDER BY qsos.COL_CALL ' . $sortorder[1]; break;
	case 4: $order = 'ORDER BY qsos.COL_MODE' .  $sortorder[1] . ', qsos.COL_SUBMODE ' . $sortorder[1]; break;
	case 7: $order = 'ORDER BY qsos.COL_BAND ' . $sortorder[1] . ', qsos.COL_SAT_NAME ' . $sortorder[1]; break;
	case 15: $order = 'ORDER BY qsos.COL_COUNTRY ' . $sortorder[1]; break;
	case 16: $order = 'ORDER BY qso.COL_STATE ' . $sortorder[1]; break;
	case 17: $order = 'ORDER BY qsos.COL_CQZ ' . $sortorder[1]; break;
	case 18: $order = 'ORDER BY qsos.COL_IOTA ' . $sortorder[1]; break;
	default: $order = 'ORDER BY qsos.COL_TIME_ON desc'; break;
}
```

AndreasK79, 2023-05-01:
And this was added to the dev branch.

closed 2023-05-03 by AndreasK79