magicbug/Cloudlog #2004: IS NOT NULL needed for eQSL upload filtering

ofadam, 2023-02-14:
If I'm not mistaken, the function for eQSL uploading is currently looking for a blank value in the `eqslqthnickname` column instead of a NULL value. Because of this, any QSOs that don't have an eQSL nickname set are flagged for uploading when they shouldn't be.

`      $this->db->where('station_profile.eqslqthnickname !=', '');` https://github.com/magicbug/Cloudlog/commit/6f8d07e1cdb3f584e37492fc5c870aec78077d36

AndreasK79, 2023-03-06:
Could you edit logbook_model.php and change eqsl_not_yet_sent function to the following:

    function eqsl_not_yet_sent() {
      $this->db->select('station_profile.*, '.$this->config->item('table_name').'.COL_PRIMARY_KEY, '.$this->config->item('table_name').'.COL_TIME_ON, '.$this->config->item('table_name').'.COL_CALL, '.$this->config->item('table_name').'.COL_MODE, '.$this->config->item('table_name').'.COL_SUBMODE, '.$this->config->item('table_name').'.COL_BAND, '.$this->config->item('table_name').'.COL_COMMENT, '.$this->config->item('table_name').'.COL_RST_SENT, '.$this->config->item('table_name').'.COL_PROP_MODE, '.$this->config->item('table_name').'.COL_SAT_NAME, '.$this->config->item('table_name').'.COL_SAT_MODE, '.$this->config->item('table_name').'.COL_QSLMSG');
      $this->db->from('station_profile');
      $this->db->join($this->config->item('table_name'),'station_profile.station_id = '.$this->config->item('table_name').'.station_id');
      $this->db->group_start();
      $this->db->where('station_profile.eqslqthnickname !=', '');
      $this->db->or_where('station_profile.eqslqthnickname is not null');
      $this->db->group_end();
      $this->db->where($this->config->item('table_name').'.COL_CALL !=', '');
      $this->db->where($this->config->item('table_name').'.COL_EQSL_QSL_SENT !=', 'Y');
      $this->db->group_start();
      $this->db->where($this->config->item('table_name').'.COL_EQSL_QSL_SENT !=', 'I');
      $this->db->or_where(array($this->config->item('table_name').'.COL_EQSL_QSL_SENT' => NULL));
      $this->db->group_end();

     return $this->db->get();
    }

If that solves your problem, I'll push it to the dev branch.

ofadam, 2023-03-06:
This seems to have worked like a charm! Thank you so much!

AndreasK79, 2023-03-07:
Thanks for testing. I've pushed the fix to dev now.

closed 2023-03-07 by AndreasK79

ofadam, 2023-03-08:
Whoops! Apparently I spoke too soon. While this fix stopped QSOs without an eQSL nickname set, it also seems to be preventing QSOs *with* an eQSL nickname set from being queued for upload.