magicbug/Cloudlog #2005: Add QSO DXCC Summary Slow and data out of columns

magicbug, 2023-02-16:
This might just be affecting me but I'm finding that with the DXCC Summary on the Add QSO page, not only is it slowing down the population of the QSO fields that I'm also getting data outside of the table.

[image]

AndreasK79, 2023-02-16:
Not seeing that problem with the table here. 

I guess everything comes with a cost when doing business with the database, but perhaps we can change it a little bit so you need to click a button or something for it to load?

magicbug, 2023-02-17:
Maybe if we can delay it to run last, not sure where in the chain, its happening need to check

I'm wondering if unknowns is due to me turning off some bands?

AndreasK79, 2023-02-18:
The call is here: qso.js:411 getDxccResult(result.dxcc.adif, convert_case(result.dxcc.entity));

As for deactivated bands, it seems you nailed that one. I deactivated one band with QSOs and I got a similar result to the one you posted.

AndreasK79, 2023-02-18:
Pushed fix for the table.

magicbug, 2023-02-18:
Push fixed the table issue thanks! will look at the js call

closed 2023-02-20 by magicbug