magicbug/Cloudlog #2012: Analytics > Timeline > Award DXCC > Show > Show QSO's doesn't show pop up with QSO's

PD0RZH, 2023-02-22:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
Analytics > Timeline > Award DXCC > Show > Show QSO's doesn't show pop up with QSO's

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to Analytics > Timeline > Award DXCC > Show > Show QSO's 
2. Verify that pop up doesn't show up

**Expected behaviour**
Like with the other choices a pop up showing QSO's should be there

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: MAC Ventura 13.1
 - Browser Safari
 - Version 16.2

AndreasK79, 2023-02-22:
It seems I broke something when I updated it for VUCC timeline. I'll take a look.

AndreasK79, 2023-02-22:
Pushed fix to dev branch.

closed 2023-02-22 by AndreasK79