magicbug/Cloudlog #2013: Countries breakdown shows clickable link to confirmed but does not lead to anything

PD0RZH, 2023-02-22:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
Countries breakdown shows clickable link to confirmed but does not lead to anything

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to cloudlog/index.php
2. Take a look half way on the right side
3. Click Confirmed (in blue)
4. Nothing happens

**Expected behaviour**
A link to a view where Confirmed QSO will be showed

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: Ventura
 - Browser Safari

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

magicbug, 2023-02-22:
Because its a roll over tooltip.

closed 2023-02-22 by magicbug

PD0RZH, 2023-02-22:
Hi Peter,

Ah for me it’s confusing since the word gets underlined when hovering thus
suggesting a link ( well in my view anyway)

But thanks for explanation.

Keep up the good work.

( oh by the way in your 2.4 announcement you referred to Awards > Timeline
and Awards > Continents. Awards should have been Analytics. )

Greetings from the Netherlands

Rob PD0RZH

soakland5, 2023-02-25:
Hi Rob, I know this is off topic, but I have been trying to get your ADIFPUSH to work. I installed DOT.net but don't know where to go from here. I also copied the .dll you created and put it in a directory but couldn't point to it. Can you give me an idea on how to do it? I am running Windows 11.