magicbug/Cloudlog #2020: [Feature Request] Export in Contest Formats (Cabrillo, EDI)

Stiles96, 2023-03-02:
It will be nice, if you can export the contest log in different formats for contest log upload.

Today I use some adif to ... converter after adif export from cloudlog.

AndreasK79, 2023-03-10:
@magicbug what about that branch you started on a few years back? Time to revive it?

magicbug, 2023-03-10:
Yep, it just needs intergrated in i'm fairly sure the format is ok

m0urs, 2023-04-23:
Just tested the Cabrillo log and I have a question if the export is really correct. In this contest there is only an exchange but no serial number. If I export into Cabrillo file there is a "000" for the serial and then followed by the exchange.

I would have said that this "000" should NOT be part of the export?

Sample:

QSO:  14169 PH 2023-04-23 1509 DG3NAB         59 000    JO40 RZ3B           59 000    KO85 0
QSO:  14153 PH 2023-04-23 1517 DG3NAB         59 000    JO40 UA1ANA         59 000    LO22 0
QSO:  14169 PH 2023-04-23 1523 DG3NAB         59 000    JO40 RZ3B           59 000    XXXXX 0
QSO:  14120 PH 2023-04-23 1532 DG3NAB         59 000    JO40 UA4WAU         59 000    LO66 0
QSO:  14159 PH 2023-04-23 1535 DG3NAB         59 000    JO40 RA9AU          59 000    MO04 0
QSO:  14162 PH 2023-04-23 1540 DG3NAB         59 000    JO40 RT4P           59 000    LO36 0
QSO:  14178 PH 2023-04-23 1543 DG3NAB         59 000    JO40 ES1BH          59 000    KO29 0
QSO:  14181 PH 2023-04-23 1545 DG3NAB         59 000    JO40 RA9V           59 000    NO34 0
QSO:  14177 PH 2023-04-23 1554 DG3NAB         59 000    JO40 R1BAC          59 000    SP14 0
QSO:  14190 PH 2023-04-23 1556 DG3NAB         59 000    JO40 RW9DX          59 000    MO06 0

AndreasK79, 2023-04-26:
Thanks for testing. That doesn't look right. I checked here, and the serial fields aren't set in the database.  That means the export pads them as zero, when they should be hidden.

I'll make a fix for that.

AndreasK79, 2023-04-26:
@m0urs I pushed a fix to the dev branch now. Please test it out.

m0urs, 2023-04-27:
@AndreasK79 Das passt so aus meiner Sicht. Vielen Dank. Jetzt ist mir aber beim Testen eben noch was anderes aufgefallen beim Contest Logging. Da mach ich ein eigenes Issue für auf.

closed 2023-05-05 by magicbug