magicbug/Cloudlog #2022: Save last Exchange also in other modes

Stiles96, 2023-03-04:
**Describe the bug**
In the contest mode, the last entered exchange (last contest) will also be saved, if the contest don't contain an exchange.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to Contest Log
2. Add Log with Exchange
3. Set Contest to other mode (in my case: Serial + Grid)
4. Log
5. See also the exchange in the log.

**Expected behaviour**
Exchange will not be saved, if not selected in the drobdown.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Win 11
 - Browser Firefox
 - Version 110

AndreasK79, 2023-04-05:
Nice catch. This was something I never thought about when this was written. I need to add a check for exchangetype and set columns accordingly. Now it just checks if exchangetype exists and sets all columns.

AndreasK79, 2023-04-06:
@Stiles96 I pushed a fix to the dev branch now. Should fix the issue with logging according to exchangetype.

closed 2023-04-11 by AndreasK79